raetha/wyzesense2mqtt #8: Bridge CLI Tool warns Flake8 lint test "C901 too complex"

raetha, 2020-03-16:
`./wyzesense2mqtt/bridge_tool_cli.py:42:1: C901 'main' is too complex (23)`

I looked at this one and am not sure how to go about removing the warning. Could use help here if anyone who sees this is good enough at Python to suggest how to "resolve" the warning.

dale3h, 2020-07-18:
Can you check this again? In Python 3.7.3 it does not appear to warn about this:

```
pi@wyzesense2mqtt:/etc/wyzesense2mqtt $ python3 --version
Python 3.7.3
pi@wyzesense2mqtt:/etc/wyzesense2mqtt $ flake8 .
./wyzesense2mqtt/wyzesense2mqtt.py:383:80: E501 line too long (83 > 79 characters)
./wyzesense2mqtt/wyzesense.py:9:1: F401 'argparse' imported but unused
./wyzesense2mqtt/wyzesense.py:15:1: E302 expected 2 blank lines, found 1
./wyzesense2mqtt/wyzesense.py:21:1: E302 expected 2 blank lines, found 1
./wyzesense2mqtt/wyzesense.py:24:1: E305 expected 2 blank lines after class or function definition, found 1
./wyzesense2mqtt/wyzesense.py:24:10: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:25:11: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:27:1: E302 expected 2 blank lines, found 1
./wyzesense2mqtt/wyzesense.py:30:1: E302 expected 2 blank lines, found 1
./wyzesense2mqtt/wyzesense.py:35:16: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:36:16: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:37:16: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:38:16: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:39:22: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:40:26: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:43:14: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:46:20: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:47:27: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:48:24: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:49:22: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:50:22: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:51:19: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:52:25: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:53:24: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:56:24: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:57:23: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:58:21: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:59:21: E221 multiple spaces before operator
./wyzesense2mqtt/wyzesense.py:61:36: E251 unexpected spaces around keyword / parameter equals
./wyzesense2mqtt/wyzesense.py:61:38: E251 unexpected spaces around keyword / parameter equals
./wyzesense2mqtt/wyzesense.py:71:80: E501 line too long (85 > 79 characters)
./wyzesense2mqtt/wyzesense.py:73:80: E501 line too long (92 > 79 characters)
./wyzesense2mqtt/wyzesense.py:85:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:92:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:138:80: E501 line too long (90 > 79 characters)
./wyzesense2mqtt/wyzesense.py:146:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:150:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:160:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:191:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:216:80: E501 line too long (88 > 79 characters)
./wyzesense2mqtt/wyzesense.py:223:1: E302 expected 2 blank lines, found 1
./wyzesense2mqtt/wyzesense.py:229:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:231:80: E501 line too long (81 > 79 characters)
./wyzesense2mqtt/wyzesense.py:233:80: E501 line too long (90 > 79 characters)
./wyzesense2mqtt/wyzesense.py:235:80: E501 line too long (84 > 79 characters)
./wyzesense2mqtt/wyzesense.py:238:1: E302 expected 2 blank lines, found 1
./wyzesense2mqtt/wyzesense.py:251:80: E501 line too long (84 > 79 characters)
./wyzesense2mqtt/wyzesense.py:265:80: E501 line too long (118 > 79 characters)
./wyzesense2mqtt/wyzesense.py:267:80: E501 line too long (87 > 79 characters)
./wyzesense2mqtt/wyzesense.py:287:48: E251 unexpected spaces around keyword / parameter equals
./wyzesense2mqtt/wyzesense.py:287:50: E251 unexpected spaces around keyword / parameter equals
./wyzesense2mqtt/wyzesense.py:314:9: E265 block comment should start with '# '
./wyzesense2mqtt/wyzesense.py:336:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:338:13: E265 block comment should start with '# '
./wyzesense2mqtt/wyzesense.py:347:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:349:13: E265 block comment should start with '# '
./wyzesense2mqtt/wyzesense.py:378:37: E251 unexpected spaces around keyword / parameter equals
./wyzesense2mqtt/wyzesense.py:378:39: E251 unexpected spaces around keyword / parameter equals
./wyzesense2mqtt/wyzesense.py:415:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:422:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:458:80: E501 line too long (86 > 79 characters)
./wyzesense2mqtt/wyzesense.py:463:80: E501 line too long (80 > 79 characters)
./wyzesense2mqtt/wyzesense.py:470:80: E501 line too long (104 > 79 characters)
./wyzesense2mqtt/wyzesense.py:493:9: E722 do not use bare 'except'
./wyzesense2mqtt/wyzesense.py:514:9: E306 expected 1 blank line before a nested definition, found 0
./wyzesense2mqtt/wyzesense.py:516:80: E501 line too long (92 > 79 characters)
./wyzesense2mqtt/wyzesense.py:518:1: W293 blank line contains whitespace
./wyzesense2mqtt/wyzesense.py:524:13: E303 too many blank lines (2)
[... cut ...]
./wyzesense2mqtt/wyzesense.py:547:80: E501 line too long (103 > 79 characters)
./wyzesense2mqtt/bridge_tool_cli.py:2:80: E501 line too long (99 > 79 characters)
./wyzesense2mqtt/bridge_tool_cli.py:75:80: E501 line too long (83 > 79 characters)
./wyzesense2mqtt/bridge_tool_cli.py:76:80: E501 line too long (91 > 79 characters)
./wyzesense2mqtt/bridge_tool_cli.py:85:80: E501 line too long (82 > 79 characters)
```

I am in the process of fixing the rest of the issues listed above. I will submit a single PR for everything that I am able to resolve.

raetha, 2020-07-18:
@dale3h I don't see this anymore either on my system. So either some other change resolved it inadvertently, or maybe the Flake8 definitions changed. Either way, happy to close it. In the long run, I think it would be ideal to re-write the CLI tool from scratch and make more of the service pieces modular so that the CLI can do everything except the MQTT pieces, without having to duplicate code.

closed 2020-07-18 by raetha