raetha/wyzesense2mqtt #9: Bad devices break the bridge

shauder, 2020-03-24:
**Describe the bug**
This is similiar or the same as the problem you commented on for my issue with the ha-gosense.  I finally took some time to try and migrate to your solution. I got a fresh bridge that had not been used with anything else. I got it up and running and started pairing through the bridge cli tool. I every pair seemed to go well and it seemed to detect them. At some point I notice it stops working. When I rebooted the container I get this error.  This is the same error I get with the other bridge when I tried it before using this fresh bridge.  I tried to run the fix commands on it from the bridge cli but each time I restart the docker container I get this error.  It definitely seems like I have a bad sensor and I might have an idea which one. I am unsure how to get rid of the bad sensor it added though.

```
sensor_mac: 77837280
Traceback (most recent call last):
  File "./wyzesense2mqtt.py", line 397, in <module>
    init_sensors()
  File "./wyzesense2mqtt.py", line 143, in init_sensors
    add_sensor_to_config(sensor_mac)
TypeError: add_sensor_to_config() missing 2 required positional arguments: 'sensor_type' and 'sensor_version'
```

**To Reproduce**
Steps to reproduce the behavior:
I setup the docker container then started pairing sensors through the bridge-cli tool inside the docker shell.  Eventually I noticed new ones would be added but mqtt events were not being added. Restarted the container and got the following error.

**Expected behavior**
Bad sensor with missing arguments is not added

**Screenshots**
n/a

**Desktop (please complete the following information):**
n/a

**Smartphone (please complete the following information):**
n/a

**Additional context**
n/a

raetha, 2020-03-26:
@shauder Interesting, at least if the logging is working right, it is telling you which sensor is bad (77837280). So you could try unpairing that sensor and removing any entry for it from the config file. If things are stable, the best way to remove is with an MQTT remove topic with the sensor MAC, as that will clear the config, MQTT topics, etc. But I'm not sure if it is stable enough, so some remains might stay in MQTT and HA that have to be cleaned up.

I can certainly look at adding some error handling to not having a version or type value, so it doesn't die out, but without that failure you might never realize something is bad.

As a side thing, if you didn't just replace the battery in that specific sensor, can you try that and then repairing? I've definitely noticed with mine that if the batteries are even remotely low, stuff starts to go wrong. Would be ideal if that simple.

Let me know, and sorry that I might be a bit slow to respond right now. I work in IT at a hospital, and things are fairly crazy at the moment. I'm finding myself a bit too tired to tinker by the time I get home.

raetha, 2020-03-26:
@shauder 
Ok, found a definite issue with my code. This isn't a bad sensor, there is a condition on load where if a sensor was paired through an alternate means, like the cli tool, it will get added to the config when the main script first starts, but it doesn't have any information to fill in the type or version, so I didn't pass those values. I needed to pass null values instead. Fixing this now and will push.

I'm going to create a new issue for actually having the cli tool write to the config file as well. I find personally that I like cli, but kept the two things separate other than interacting with the hub. If you did all the pairing using MQTT topics, then the main script would have handled things without issue. But you found the one place where I had something that would never work right. :)

Look for version 1.0.1 shortly. I'm going to punt the other issues and documentation updates to a newer update so I can get you fixed ASAP.

closed 2020-03-26 by raetha

shauder, 2020-03-26:
Thank you seems to be working so far.