raetha/wyzesense2mqtt #13: Add MQTT topic to reload sensor config

raetha, 2020-03-29:
To allow for easier renaming of sensors without having to restart the service or docker container, add a topic and subscription that will reload the sensor config while the service is running.

onedr0p, 2020-07-07:
Would be nice also if the docker image didn't require logging, sensor and config present when starting up the first time.

raetha, 2020-07-11:
Hi @onedr0p.

I'll take a look at if we can do anything with docker. But without a web UI to configure things, there is nothing to start in docker without the config.yaml file in place. the sensors one will just build itself, but the config file tells it how to connect to MQTT, and without that nothing would really happen. So what would you want to docker container to do if that file isn't present? Just loop looking for it until one is put in place?

On the logging front, I was mostly just experimenting with Python logging. I learned Python for this project, and so wanted to dig in more than one might when trying to do something quick. Since I got the dynamic logging using dicts working, I figured there was no reason to remove it, but also defaulted that so that no one needed to modify it if they didn't want to.

onedr0p, 2020-07-11:
I was actually thinking a web UI would be great, even if it was a white page with a few inputs and textboxes to configure devices. Maybe a project in the future if I can ever get to it... haha.

I didn't know their was a default for logging, I thought I had to include it in my volume mount.

What about a single config file for everything? Something like:

```yaml
mqtt:
  host: <host>
  port: 1883
devices:
  - name: Front Door
    mac: 'XXXXXX'
    location: Foyer
  - name: Patio Door
    mac: 'XXXXXX'
    location: Patio
```

That would make configuration much easier, I'll open up a separate issue if you think that's good :)

raetha, 2020-07-13:
@onedr0p Definitely some good ideas in here. I've already got an issue logged for a web GUI, though I don't have the skills for that, but am asking some developer friends for any framework recommendations or anything else they could help with there.

On the config files, for the moment I'd like to keep them separate. The sensors config is especially what I'd like to see editable in a web UI, but it is also dynamically built with defaults by the script. If I merge it, everything has to be ready and able to write out the complete configuration or risk breaking stuff. Right now if the sensors file blows up, things should get rebuilt (without customizations), but things will still work. If the MQTT settings get broken though, then everything grinds to a halt.

I did just add issue #28 (and the code) to automatically copy a default config.yaml and logging.yaml if they don't exist. So Docker runs will now put them in the config folder waiting to be modified as needed.

As for Docker ENVs, I'll happily take new issues for that. I want to get 1.1 released before digging into the Docker stuff more, but could look at it for 1.2. I'm thinking ENVs for the MQTT host, port, username, and password would be good, as then you could get away without ever having to touch config files for Docker. And an eventual UI to edit things after the fact would take care of the rest. And I think that's ultimately what you're looking for. I'd love you to log this one though, as having more folks involved in this has helped keep my interest going and know I'm not just tweaking things for myself. :)

I'll also definitely take a look at click. If nothing else I think that would be great for the cli tool to interact with the bridge, but to your points above might also help with some of the configuration.

Thanks again for all the feedback.

PR #15 (2020-07-17): 1.1
Add ability to reload sensor config while running (Resolves #13)
Added link to Python Validation runs
Switch to custom WyzeSensePy library (fixes #12, fixes #17)
Bump version to 1.1
Better tracking of timestamps for sensor event (resolves #21)
Add sensor invert_state option to reverse state value (resolves #23)
Fix issue with undefined SENSORS during events before SENSOR load (fixes #17)
Automatically copy sample config.yaml and logging.yaml if they don't exist (resolves #28)
Explictly cover Raspian version tested with
Add version based Docker image support (Docker Hub Autobuild) (resolves [... cut]
closed 2020-07-17 by raetha