raetha/wyzesense2mqtt #22: Config appears to be published before connected, and connection takes ~1 minute

dale3h, 2020-06-16:
## Describe the Bug

When the wyzesense2mqtt service is started, 

### Steps To Reproduce

Steps to reproduce the behavior:

1. Start/restart `wyzesense2mqtt` service (i.e. `sudo systemctl start wyzesense2mqtt`)
2. View logs (i.e. `sudo journalctl -fu wyzesense2mqtt`)
3. Look for `Connecting to MQTT host ...` and note the timestamp
4. Look for `Connected to MQTT: return code 0` and note the difference in time

### Expected Behavior

Connection to MQTT broker should only take a second or two at most, and payloads should not be published until after connected to MQTT broker.

## Additional Context

### Installation Method

I am running `wyzesense2mqtt` on latest Raspberry Pi OS (`Raspbian GNU/Linux 10 (buster)`) using a `systemd` service. Host is a Raspberry Pi 3 Model B+.

### Related Logs

https://hastebin.com/ihuloxoxam

raetha, 2020-06-19:
This one is interesting. I've never seen on my side the MQTT connection take that long. Since the connection is async, the topics may publish in memory before the connection is established, but the MQTT library should hold those and send them once the connection is up, so there shouldn't be an impact to that timing. I would like to see if we can speed up the MQTT connection though.

I assume you're using Mosquitto, is it running on the same host, or different host?

dale3h, 2020-07-07:
Yes, I am using mosquitto. It is running on the same hypervisor/machine, but under a different VM.

The queue topics seem to be successfully sent once the connection is finally established.

Could this be the result of me having as many sensors as I do? Is there any way for me to test it without publishing everything before the connection is established? This would narrow it down to whether or not it's being delayed by the queue.

raetha, 2020-07-12:
I'm using docker instead of VMs, but similar type of set up. So shouldn't be a massive difference there.

Looking at the logs you posted originally, I see a big time gap between the last publish and the mqtt client connection, so I'm thinking it shouldn't be the extra publishes early, but I'm not hugely knowledgeable on the Paho MQTT library and how it works. It may be that having a large queue of topics to send before the connection is established does cause some delay.

If you want to test, just comment out these lines in the init_sensors function. This will skip the publishing of the discovery topics on start. Note that could have adverse affects long term though as I think it's the only time existing sensors are republished, so if those topics ever expire or disappear, they wouldn't be recreated.
```
    for sensor_mac in SENSORS:
        if (valid_sensor_mac(sensor_mac)):
            send_discovery_topics(sensor_mac)
```

Let me know what you find. We could also go the route of forming new connections to mqtt only when needed and publish in realtime, but that will compound any lingering delays for every single message instead of just on initial connection.

raetha, 2020-07-13:
The other possible thing I found that might help here is to add true threading support for the bridge and mqtt sides of things, rather than just using the built in functions. My worry is that gets well out of my wheelhouse, but I think could be worth testing. Then those two threads could be started as soon as the config is loaded, but before sensors and other things. We could then easily track when the mqtt connection tries to start, and when it succeeds. I'm just not sure it's worth the complexity at this point. At least not without someone contributing who really understands threading and how best to implement it.

dale3h, 2020-07-18:
The time gaps have shifted, but are still present between **connecting** and **connected**:

```
Jul 17 23:32:35 wyzesense2mqtt python3[3115]: Connecting to MQTT host XXXXXXXX
...(1s)...
Jul 17 23:32:36 wyzesense2mqtt python3[3115]: Reading sensors configuration...
...(52s)...
Jul 17 23:33:28 wyzesense2mqtt python3[3115]: Publishing discovery topics for 77800DFB
...(0s)...
Jul 17 23:33:28 wyzesense2mqtt python3[3115]: Connected to MQTT: return code 0
```

More log output available here: https://hastebin.com/agupimarif

dale3h, 2020-07-18:
I commented out the sending of discovery topics in `init_sensors` and the delay is still present. I am adding in some extra debug output that might help me figure out where the delay is happening.

dale3h, 2020-07-18:
I am no longer having this issue after discovering that the 26th sensor was one with an invalid MAC address. I was NOT able to remove the sensor by publishing `00000000` to `wyzesense2mqtt/remove`, but trying to re-pair the sensor using `wyzesense2mqtt/scan` "kind of" worked. I say "kind of" because it crashed wyzesense2mqtt altogether, and required a restart of the service (which threw the `NameError: name 'SENSORS' is not defined` [see #17] the first couple of times it attempted to start). Here is the log output from attempting to pair the invalid sensor:

```
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: Scan result: ('\x00\x00\x00\x00\x00\x00\x00\x00', 1, 16)
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: sensor_mac:
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: Unpairing bad MAC:
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: LOG: time=2020-07-18T02:51:31.981000, data=b'250000000000000000'
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: LOG: time=2020-07-18T02:51:31.981000, data=b'250000000000000000'
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: Clearing sensor topics
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: Invalid sensor found:
Jul 18 02:51:59 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
Jul 18 02:52:00 wyzesense2mqtt python3[5183]: Connected to MQTT: return code 0
Jul 18 02:52:00 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
Jul 18 02:52:01 wyzesense2mqtt python3[5183]: Connected to MQTT: return code 0
Jul 18 02:52:01 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
Jul 18 02:52:02 wyzesense2mqtt python3[5183]: Connected to MQTT: return code 0
Jul 18 02:52:02 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
Jul 18 02:52:03 wyzesense2mqtt python3[5183]: Connected to MQTT: return code 0
Jul 18 02:52:03 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
...this continued every second until it was restarted...
Jul 18 02:52:29 wyzesense2mqtt python3[5183]: Connected to MQTT: return code 0
Jul 18 02:52:29 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
Jul 18 02:52:30 wyzesense2mqtt python3[5183]: Connected to MQTT: return code 0
Jul 18 02:52:30 wyzesense2mqtt python3[5183]: Disconnected from MQTT: return code 1
Jul 18 02:52:30 wyzesense2mqtt systemd[1]: Stopping wyzesense2mqtt...
Jul 18 02:52:30 wyzesense2mqtt systemd[1]: wyzesense2mqtt.service: Main process exited, code=killed, status=15/TERM
Jul 18 02:52:30 wyzesense2mqtt systemd[1]: wyzesense2mqtt.service: Succeeded.
Jul 18 02:52:30 wyzesense2mqtt systemd[1]: Stopped wyzesense2mqtt.
```

tl;dr: Delay was caused by a sensor that was previously paired with an invalid MAC. Remove the invalid sensor by pairing it again. Restart wyzesense2mqtt.

closed 2020-07-18 by dale3h

raetha, 2020-07-18:
@dale3h Nice find. I've been trying to add more error handling for the invalid MAC issue, including auto deletion if it tries to pair one in the first place, however I don't think there is a good way to detect if one already exists. The CLI tool does have a fix function to unpair that way.

What would be interesting to see is if you know which sensor that is, and it still has a low battery and tries to pair with the bad MAC, whether or not the auto-unpair works. If you do have that scenario still, can you try doing a scan and pairing it and see what the logs give you? Or maybe that is what you did, in which case just confirm. Then we just need to solve the fact that it nuked your service rather than gracefully deleting and moving on.