raetha/wyzesense2mqtt #23: Add invert_state config option to reverse on/off state reporting

jfarmer08, 2020-06-25:
Add the ability to invert the state of the sensors.

raetha, 2020-06-26:
@jfarmer08, can you give me some more information on what you are looking for here? Specifically:
- Confirming that you want HA to see the sensor as closed when it is reported as open?
- Are you looking to have all sensors invert their state? Or just specific ones?
- Example of use case for this functionality

I have no issues with the request, it will just add quite a bit more complexity to some of the functions, and determining how to track the setting. Currently I'm replicating what other projects are doing (and I believe the HA standard), so want to understand better what you are looking for and why to make sure it doesn't cause issues down the line.

Thanks!

jfarmer08, 2020-06-26:
@raetha I am using the Wyze sensor with a water sensor on it. And so now it reports open all the time when no water is present and when water is present it reports closed. So 99.9% of the time it reports open. So when i set the class to moisture it reports to HA as Wet. 

So with the thought of inverted state i would be able to change the state before its reported to HA. Yes i could create a template for this, but I think this is a much cleaner solution. 

[DIY Wyze Water Sensor](https://www.youtube.com/watch?v=7gNqCTPAiP0)

So with the thought in mind we could just use the sensors.yaml file. You can default the invert to false during creation of the file and maybe make it optional, so if invert is not in there just assume false. Hopefully this is what you were looking for. 

```
'AAAAAAAA':
  class: moisture
  name: Sump Pump
  invert: True
```

raetha, 2020-06-26:
@jfarmer08 that's perfect information, thanks. And I like the per-sensor option. I'm going to do a little digging through the HA data though, as a nicer way to handle this might be to instead recognize moisture as a sensor type and treat that the way HA expects it. Then all you would need is to set the class and the rest of the logic would be automatic.

I'll try to dig in this weekend and at least get something in the devel branch and builds by Sunday for you to take a look at.

jfarmer08, 2020-06-26:
@raetha As far as HA is concern its a binary sensor. So setting the Device class to Moisture is all that is need. From There HA now shows it as dry or wet based on the State. So for the logic we just need a boolean(invert) in the config. And before sending the state to MQTT check this boolean to see if it needs to be inverted. 

I would be glad to help with this, but I would require help with setting you a Dev environment. 

[image]

raetha, 2020-06-28:
@jfarmer08 just committed some changes for this to the devel branch. Can you either switch docker to the devel container when you get a chance (will take 30 minutes or so to build), or grab a replacement copy of the script from the devel branch here.

Decided I liked your invert_state sensors config option. The more I thought about it, I realized that moisture sensors work the same way, but someone could configure a contact sensor the reverse of how you've done if they really wanted to, and thus hard coding the change to moisture might not work for everyone, whereas being able to set it in config does allow for that.

Just add an "invert_state" value of "true" to your sensors.yaml after switching to the script, and let me know if that works as expected. You'll need to trigger the sensor for the MQTT to update.

jfarmer08, 2020-06-29:
@raetha I have switch over to the devel branch. And so far everything looks good.

raetha, 2020-06-29:
Great to hear! I've got a little more documentation updates I want to do relating to this, and then see if I want to sneak anything else into 1.1. Hopefully I'll get a chance to lock it down and push out by the end of the week. Then you can switch back to the stable branch if you desire. Though also happy to have folks stay on devel and let me know if anything breaks with future changes.

dale3h, 2020-07-01:
I am in a situation where these changes will be helpful to me as well. I have an unmodified Wyze Sense contact sensor inside of my doorbell chime enclosure, so that it reports closed when someone rings the doorbell, but otherwise open. I didn't think it'd be worth requesting as a feature, so I just setup a Template Binary Sensor to invert it inside of HA itself. I'll update to latest `devel` and give the changes a whirl!

As for staying on the `devel` branch, I might end up doing that. I have a watchdog timer setup to notify me whenever a wyzesense2mqtt topic hasn't been seen in 1h. I get a few false positives, but it's generally through the night or when we are away from the home for a while.

dale3h, 2020-07-07:
After a week of use, this specific feature seems to be working great for me. Thank you again for the addition!

jfarmer08, 2020-07-07:
Yes the feature seems to be working good.

raetha, 2020-07-12:
Thank you both for confirming. Waiting to solidify a couple more things and then will push this to master in case you want to go back to the "stable" branch. :)

commit a9761f14 (2020-06-28): Add feature to invert the state values for a sensor (resolves #23)
PR #15 (2020-07-17): 1.1
Add ability to reload sensor config while running (Resolves #13)
Added link to Python Validation runs
Switch to custom WyzeSensePy library (fixes #12, fixes #17)
Bump version to 1.1
Better tracking of timestamps for sensor event (resolves #21)
Add sensor invert_state option to reverse state value (resolves #23)
Fix issue with undefined SENSORS during events before SENSOR load (fixes #17)
Automatically copy sample config.yaml and logging.yaml if they don't exist (resolves #28)
Explictly cover Raspian version tested with
Add version based Docker image support (Docker Hub Autobuild) (resolves [... cut]
closed 2020-07-17 by raetha