raetha/wyzesense2mqtt #24: Version docker images

onedr0p, 2020-07-06:
Hi! Thanks so much for creating this image, I was all set to create my own from the upstream project before I found this.

I am curious if you plan to support version tags on the docker images? This would be great incase we need to rollback to a previous version.

For example:

current version would be the same image with different tags:
```
raetha/wyzesense2mqtt:v1.1.0
raetha/wyzesense2mqtt:latest
```

previous versions:
```
raetha/wyzesense2mqtt:v1.0.9
raetha/wyzesense2mqtt:v1.0.8
...
```

raetha, 2020-07-07:
Hi Devin (@onedr0p), I'd actually originally set things up with releases when I was first figuring out Docker Hub, but removed it for some reason. I honestly can't remember my logic, and yours makes a ton of sense, so I'll get them back in place. I'll update here once that is complete.

raetha, 2020-07-07:
Ok, I believe I've added the required configuration in Docker, unfortunately it looks like it may not start generating them until the next release, so hopefully I'll know as soon as I can polish 1.1 and get that pushed. I'm hoping by this weekend. I'll leave this open until I confirm that they are generating going forward.

onedr0p, 2020-07-07:
Thanks @raetha !

PR #15 (2020-07-17): 1.1
Add ability to reload sensor config while running (Resolves #13)
Added link to Python Validation runs
Switch to custom WyzeSensePy library (fixes #12, fixes #17)
Bump version to 1.1
Better tracking of timestamps for sensor event (resolves #21)
Add sensor invert_state option to reverse state value (resolves #23)
Fix issue with undefined SENSORS during events before SENSOR load (fixes #17)
Automatically copy sample config.yaml and logging.yaml if they don't exist (resolves #28)
Explictly cover Raspian version tested with
Add version based Docker image support (Docker Hub Autobuild) (resolves [... cut]
closed 2020-07-17 by raetha