raetha/wyzesense2mqtt #25: Add friendly name to event MQTT message

dennyfmn, 2020-07-08:
I wanted to be able to parse the sensors by their friendly name so I added a couple of lines to wyzesense2mqtt.py

# Process event
def on_event(WYZESENSE_DONGLE, event):
    if (valid_sensor_mac(event.MAC)):
        if (event.Type == "state"):
            LOGGER.info(f"State event data: {event}")
            (sensor_type, sensor_state, sensor_battery, sensor_signal) = event.Data
            sensor_name = SENSORS[event.MAC]['name']    <<<---------
            event_payload = {
                'available': True,
                'name': sensor_name, <<<------------
                'mac': event.MAC,

This works for me. Please consider adding it to the main code.

dale3h, 2020-07-09:
I like this idea, but I would suggest that it be configurable via `config.yaml`.

dennyfmn, 2020-07-09:
Thank you for turning this into a proper pull request; I'm not facile with GIT. I appreciate your quick and well thought out response.

raetha, 2020-07-11:
@dennyfmn thanks for the good idea here, and @dale3h thanks for the quick implementation. I submitted a few requested tweaks on the PR, but otherwise will be happy to pull this in this weekend. And I think I'll wrap 1.1 so that this is officially released as well.

dale3h, 2020-07-16:
This issue can be closed as per #26 and 7684bffc309895d1f29fd1d17b0f8473fe935577.

PR #15 (2020-07-17): 1.1
Add ability to reload sensor config while running (Resolves #13)
Added link to Python Validation runs
Switch to custom WyzeSensePy library (fixes #12, fixes #17)
Bump version to 1.1
Better tracking of timestamps for sensor event (resolves #21)
Add sensor invert_state option to reverse state value (resolves #23)
Fix issue with undefined SENSORS during events before SENSOR load (fixes #17)
Automatically copy sample config.yaml and logging.yaml if they don't exist (resolves #28)
Explictly cover Raspian version tested with
Add version based Docker image support (Docker Hub Autobuild) (resolves [... cut]
closed 2020-07-17 by raetha