raetha/wyzesense2mqtt #29: Add Neos Leak Sensor support

raetha, 2020-07-13:
Per https://github.com/HclX/WyzeSensePy/pull/13 Neos is using a similar bridge and technology, but has an additional leak sensor that Wyze does not yet have. Since this project forked the WyzeSensePy library, the functionality can be built into this version of the library and WyzeSense2MQTT. Assuming user validation after the small number of changes, we can then document support for Neos as well as Wyze.

jellybob, 2020-07-22:
Hi there, I'd love to see this change merged, particularly if it enables Home Assistant support, I'll try to pull it downstream here since this version seems like a better target to be able to do something useful with the code. In my testing door/window and motion sensors just work out of the box (although they do have a different model number if you're making use of those), while the leak sensors need a bit of hacking around, particularly around getting humidity readings out of them.

raetha, 2020-07-22:
@jellybob, hoping to merge this into the devel branch by this weekend at the latest. If you see the PR that @dale3h did, we just need to get some additional changes added to add the support to the WS2MQTT script, rather than just the underlying fork of WyzeSensePY. I'm thinking I'll have some time on Friday to flesh that out if Dale doesn't beat me to it.

As far as the differences, I don't think the model numbers should matter, as long as the USB communications are the same. My only real concern is that if Wyze and Neos go different directions with the USB stuff, accommodating both in a single project would add some complexity, but that might just be the time to fork this into a Neos specific version if it becomes to cumbersome. Alternately, with more community involvement, I'm happy to let this expand to cover both in a single implementation.

On the leak sensors and humidity readings, is that working in the code sample you had originally provided? Or do we need to look at adding more support for that?

dale3h, 2020-07-22:
@raetha I have already started on the changes to `wyzesense2mqtt.py`. I will try to finish them this evening and get them committed to the PR.

For the humidity readings, I remember reading something about him saying that the reading was vastly different between what he was able to pull with the library vs what the app said. I believe the app was saying something around 66% and the reading was 83%. I even tried converting those between bases and couldn't find any reasonable correlation.

raetha, 2020-07-23:
That's too bad, be nice to add the extra data point, but if it's unreliable (or not translatable), not much value in presenting bad data.

dale3h, 2020-07-23:
@raetha I recommend you taking a look at his original code to see if you can find something I may have missed. Of course, it would be extremely helpful if either of us had a Neos Smart Leak Sensor to test out. I've setup some search alerts on eBay in hopes that one may become available eventually. If not, I may see if someone in the Home Assistant Discord might be able to send one over our way.

jellybob, 2020-07-23:
@dale3h @raetha I should be able to get some sensors shipped out to you, drop me an email on jon.wood@neos.co.uk and I'll try to get it sorted. I can also say there's no intention to change the USB protocol.

jellybob, 2020-07-23:
@raetha leak and temperature readings looked pretty solid to me, humidity was wildly out - the value I'm reading for it is as far as I can see the only place it could be, but I'm not sure what the encoding is. We're post-launch of the new devices in the UK at the moment, but once things calm down a little I'll see if I can get some details on that from the OEM.

raetha, 2020-07-25:
@jellybob Really appreciate the kind offer, and it's fantastic that you are actually connected to the company, gives us a much higher chance of getting this working well. I've just shot you an email as well.

I'm looking through what @dale3h has done so far, and between the two of us will get something pulled into the devel branch ASAP. If you have the ability to help test, that would be great as well, but I'm confident we can cycle through this and get at least a basic set of things working quickly. Then with help get things really solid. My background on hardware communications is non-existent, but I'm happy to beat on things, and with your inside knowledge we should be good.

raetha, 2020-07-25:
@jellybob One other thing we are tossing around on the PR itself is how to handle the temperature. In your sample code we see that the event type is 0xE8. Again, knowing little about the hardware side, is that event type simply a different one for a temperature reading, and each time the temperature changes it will send that event? Or is there some other trigger that is occurring there, like a temperature warning. I also see you flagged it as type "leak:temperature", is it actually associated with the leak aspect, or is it really just a combined sensor that includes both a leak and temperature sensor? Want to make sure we're passing it through, particularly for Home Assistant, as the right type of thing. AKA, it's own sensor, or just an attribute associated with the leak sensor.

PR #41 (2021-02-11): 1.2
Version bump to 1.2
Update bug report template to use markdown (@dale3h)
Resolve most flake8 issues (@dale3h)
Bridge connection retries should now continue when an IOError is triggered (#17)
Add vim to docker image (Resolves #39)
Add Neos Leak Sensor support (Resolves #29) - In Progress
Handle not having a sensors.yaml file on startup (Fixes #44, Reference PR #45 @AK5nowman)
Logging enhancements (Reference PR #45 @AK5nowman)
Add support to disable HASS discovery topics (Reference PR #45 @AK5nowman)
Refactored configuration initialization to be self-updating for new features
Removing a sens [... cut]
closed 2021-02-11 by raetha