raetha/wyzesense2mqtt #37: Detect sensors already paired by HACS Wyzesense integration

SamuelCarson, 2020-07-28:


SamuelCarson, 2020-07-28:
Will this component detect sensors already paired by the Wyzesense integration or will they need to be unpaired and re-paired

raetha, 2020-07-28:
Hi @SamuelCarson, it should detect all sensors paired with the bridge, and automatically add them to the sensors.yaml file when it runs. They won't have intuitive names at that point, but you can then rename them in the config file or directly in Home Assistant after they are discovered.

SamuelCarson, 2020-09-06:
Bridge tool shows:
15 sensors paired:
	Sensor: 779C04F4
	Sensor: 779C04C9
	Sensor: 779D582D
	Sensor: 7788277B
	Sensor: 7786E6CD
	Sensor: 77881982
	Sensor: 77882771
	Sensor: 77882895
	Sensor: 77882CDF
	Sensor: 77882C6B
	Sensor: 778828A0
	Sensor: 77869E12
	Sensor: 77882724
	Sensor: 77882C9A
	Sensor: 7786E6B8
Home Assistant shows these as Unavailable and wyzesense2mqtt has NOT updated the sensors.yaml file. I  am completely stumped

SamuelCarson, 2020-09-07:
Figured it out. There was no /wyzesense2mqtt/logs directory.
Several changes to linux install are necessary, especially if running on a raspberry pi. Complete updated instructions are in the following file
[Install wyzesense2mqtt.txt](https://github.com/raetha/wyzesense2mqtt/files/5184938/Install.wyzesense2mqtt.txt)

raetha, 2020-11-28:
@SamuelCarson really appreciate the updated docs. I've incorporated those changes into the devel instructions, and they will update fully on the 1.2 release.

PR #41 (2021-02-11): 1.2
Version bump to 1.2
Update bug report template to use markdown (@dale3h)
Resolve most flake8 issues (@dale3h)
Bridge connection retries should now continue when an IOError is triggered (#17)
Add vim to docker image (Resolves #39)
Add Neos Leak Sensor support (Resolves #29) - In Progress
Handle not having a sensors.yaml file on startup (Fixes #44, Reference PR #45 @AK5nowman)
Logging enhancements (Reference PR #45 @AK5nowman)
Add support to disable HASS discovery topics (Reference PR #45 @AK5nowman)
Refactored configuration initialization to be self-updating for new features
Removing a sens [... cut]
closed 2021-02-11 by raetha