raetha/wyzesense2mqtt #38: Gateway fails to start

ha14937, 2020-08-09:
## Describe the Bug

Apologies in advance if this is not a bug  - but I'm stuck here. I'm getting a "Failed to start WyzeSense to MQTT Gateway" error when I start, then check the status of the service. Was really hoping to get this working, as attaching the bridge to the VM that runs my HA instance has become unstable.

So I am trying to get this set up on a spare Pi 3B+ and avoid Docker if possible so the Linux Systemd instructions looked like exactly what I need.

### Steps To Reproduce

I followed the Linux Systemd instructions - with the only difference being the location I stored the files (/home/pi/wyzesense2mqtt/wyzesense2mqtt)

I did mess up a few times, but deleted everything and started over - and the files all appear to be in the right places now.

### Expected Behavior

I expected the service to start.

### Desktop

- OS: Rasbian
- Browser: N/A (CLI)
- Version: Buster

### Smartphone

N/A

## Additional Context

**Here is the result of the check status command in the CLI:**

```
* wyzesense2mqtt.service - WyzeSense to MQTT Gateway
   Loaded: loaded (/etc/systemd/system/wyzesense2mqtt.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2020-08-08 21:10:28 EDT; 29s ago
     Docs: https://www.github.com/raetha/wyzesense2mqtt
  Process: 7696 ExecStart=/home/pi/wyzesense2mqtt/wyzesense2mqtt/service.sh (code=exited, status=203/EXEC)
 Main PID: 7696 (code=exited, status=203/EXEC)

Aug 08 21:10:28 raspberrypi systemd[1]: wyzesense2mqtt.service: Service RestartSec=100ms expired, scheduling restart.
Aug 08 21:10:28 raspberrypi systemd[1]: wyzesense2mqtt.service: Scheduled restart job, restart counter is at 5.
Aug 08 21:10:28 raspberrypi systemd[1]: Stopped WyzeSense to MQTT Gateway.
Aug 08 21:10:28 raspberrypi systemd[1]: wyzesense2mqtt.service: Start request repeated too quickly.
Aug 08 21:10:28 raspberrypi systemd[1]: wyzesense2mqtt.service: Failed with result 'exit-code'.
Aug 08 21:10:28 raspberrypi systemd[1]: Failed to start WyzeSense to MQTT Gateway.
```

**Here is my wyzesense2mqtt.service file, that was copied over to** `/etc/systemd/system/`
```
[Unit]
Description=WyzeSense to MQTT Gateway
Documentation=https://www.github.com/raetha/wyzesense2mqtt
After=network.target

[Service]
Type=simple
WorkingDirectory=/home/pi/wyzesense2mqtt/wyzesense2mqtt
ExecStart=/home/pi/wyzesense2mqtt/wyzesense2mqtt/service.sh
Restart=always

[Install]
WantedBy=multi-user.target
```

**Here is my config file at the absolute path of**  `/home/pi/wyzesense2mqtt/wyzesense2mqtt/config.config.yaml`:

```
mqtt_host: 192.168.1.237
mqtt_port: 1883
mqtt_username: my_mqtt_username
mqtt_password: my_mqtt_password
mqtt_client_id: wyzesense2mqtt
mqtt_clean_session: false
mqtt_keepalive: 60
mqtt_qos: 0
mqtt_retain: true
self_topic_root: wyzesense2mqtt
hass_topic_root: homeassistant
publish_sensor_name: true
usb_dongle: auto
```

jeffrutland, 2020-08-09:
I just struggled through getting this running on a pi myself.  I tried both the systemd approach, and the docker approach - I eventually got docker working.

may be an obvious question, but you list your file with the name `config.config.yaml`.  is that the actual file name?  it should be `config.yaml`.

only starting here, as I've found that some errors I had getting things rolling weren't the most helpful.

ha14937, 2020-08-09:
@jeffrutland appreciate the reply, and glad I'm not the only one who ran into troubles. 

RE: the config.config.yaml - this was a typo. The absolute path to the config.yaml file is  `/home/pi/wyzesense2mqtt/wyzesense2mqtt/config/config.yaml`. 

[image]

Step 4 has you run this command: `cp samples/config.yaml config/config.yaml` to copy the confi.yaml over to a folder named "config", and name the copy of the file config.yaml. I thought it was odd/potentially problematic, because A) instructions never tell you to create a new folder called config to place the config.yaml in and B) this puts the config.yaml one directory lower/within the "WorkingDirectory" - though I'm not sure this is causing the problem, just grasping at straws :).

- One thing I wasn't 100% about were the paths in the `wyzesense2mqtt.service` file. I assumed that: 

1. "**WorkingDirectory**" meant the folder that contained the various Python files. In my case this is `/home/pi/wyzesense2mqtt/wyzesense2mqtt` and
2. the **ExecStart** needed an absolute path to the service.sh
[image]

dale3h, 2020-08-09:
I originally had issues getting it setup in Docker, so I switched to Linux Systemd as well. But, I then had issues getting that setup, too. Here is how I currently have it working:

I have the repo cloned to `/etc/wyzesense2mqtt`, and then a symlink setup at `/opt/wyzesense2mqtt` pointing to `/etc/wyzesense2mqtt/wyzesense2mqtt`.

### File `/etc/systemd/system/wyzesense2mqtt.service`

```ini
[Unit]
Description=wyzesense2mqtt
After=network.target

[Service]
Type=simple
User=root
WorkingDirectory=/opt/wyzesense2mqtt
ExecStartPre=/bin/sleep 10
ExecStart=python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py
Restart=always
RestartSec=3

[Install]
WantedBy=multi-user.target
```

### File `/opt/wyzesense2mqtt/config/config.yaml` (`/etc/wyzesense2mqtt/wyzesense2mqtt/config/config.yaml`)

```yaml
mqtt_host: 123.123.123.123
mqtt_port: 1883
mqtt_username: my_mqtt_username
mqtt_password: my_mqtt_password
mqtt_client_id: wyzesense2mqtt-1
mqtt_clean_session: false
mqtt_keepalive: 60
mqtt_qos: 2
mqtt_retain: true
self_topic_root: wyzesense2mqtt
hass_topic_root: homeassistant
publish_sensor_name: true
usb_dongle: auto
```

### File `/opt/wyzesense2mqtt/config/sensors.yaml` (`/etc/wyzesense2mqtt/wyzesense2mqtt/config/sensors.yaml`)

```yaml
777EDD2F:
  class: door
  invert_state: false
  name: Master Shower Door
777EF86E:
  class: door
  invert_state: false
  name: Utility Room Door
777EFF2A:
  class: door
  invert_state: false
  name: Master Bathroom Door
...
```

I don't think it makes a difference (since I have `User=root` in the systemd service file), but I have `/etc/wyzesense2mqtt` and all its children owned by `pi:pi` (`sudo chown -R pi:pi /etc/wyzesense2mqtt`).

dale3h, 2020-08-09:
@ha14937 Also, based on your screenshot above, it appears that you do not yet have the `logging.yaml` and `sensors.yaml` files in your `config` directory. Please try copying both of these from the `samples` directory to see if you can at least get it to start.

@raetha I really think we need to create default config files on first startup (or at least copy the samples) if this isn't something that is already happening with the latest code. I've been away for a few weeks and haven't brushed up on the recent changes.

jeffrutland, 2020-08-09:
yes, the majority of my issues boiled down to missing / incorrect configuration.  I was actually thinking of branching and creating a PR with better error reporting when the config files are the root cause of issues.  :)

now that it’s running for me, it’s a great addition to my home assistant setup.  I’m relatively new to DIY smart home, but have been in IT for quite awhile.  I love this new adventure while being stuck at home, and the Wyze sensors are a great, cheap addition with a pi!

ha14937, 2020-08-12:
@dale3h thanks for the detailed reply. Have not got it working yet - but I think I'm closer. I have next to zero linux experience - so as I went thru the steps I took, I noted them all down, along with the CLI responses, etc. Sorry for the wall of text!

1) Cloned the repo to /etc:
```
cd /etc
git clone https://github.com/raetha/wyzesense2mqtt.git
```

2) Verified files have been cloned to /etc - all looks good to me
```
pi@raspberrypi:/etc/wyzesense2mqtt/wyzesense2mqtt $ ls
bridge_tool_cli.py  requirements.txt  samples  service.sh  wyzesense2mqtt  wyzesense2mqtt.py  wyzesense2mqtt.service  wyzesense.py
```

3) Set up a symlink to opt/wyzesense2mqtt. I have never done this, so googled a bit and came up with this, which I think worked.
`sudo ln -s /etc/wyzesense2mqtt/wyzesense2mqtt /opt/wyzesense2mqtt
`

4) Verify files show up in /opt/wyzesense2mqtt - all looks good to me, right?
```
pi@raspberrypi:/opt/wyzesense2mqtt $ cd /opt/wyzesense2mqtt
pi@raspberrypi:/opt/wyzesense2mqtt $ ls
bridge_tool_cli.py  ln  requirements.txt  samples  service.sh  wyzesense2mqtt  wyzesense2mqtt.py  wyzesense2mqtt.service  wyzesense2mqttsudo  wyzesense.py
```

5) Made sure permissions are set. I think permissions may have been part of my problem initially, but not sure.
`sudo chown -R pi:pi /etc/wyzesense2mqtt
`

6) Nav to and Edit the wyzesense2mqtt.service file
```
pi@raspberrypi:/opt/wyzesense2mqtt/config $ cd /opt/wyzesense2mqtt
pi@raspberrypi:/opt/wyzesense2mqtt $ ls
bridge_tool_cli.py  config  ln  requirements.txt  samples  service.sh  wyzesense2mqtt  wyzesense2mqtt.py  wyzesense2mqtt.service  wyzesense2mqttsudo  wyzesense.py
pi@raspberrypi:/opt/wyzesense2mqtt $ nano wyzesense2mqtt.service
```

7) In wyzesense2mqtt.service file, paste in example from @dale3h:
```
[Unit]
Description=wyzesense2mqtt
After=network.target

[Service]
Type=simple
User=root
WorkingDirectory=/opt/wyzesense2mqtt
ExecStartPre=/bin/sleep 10
ExecStart=python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py
Restart=always
RestartSec=3

[Install]
WantedBy=multi-user.target
```

8) Copy wyzesense2mqtt.service file to /etc/systemd/system/
`pi@raspberrypi:/opt/wyzesense2mqtt $ sudo cp wyzesense2mqtt.service /etc/systemd/system/
`

9) Verified file copied to /etc/systemd/system/ and new text from @dale3h appears in the service file; it did, and it does.
```
pi@raspberrypi:/opt/wyzesense2mqtt $ cd /etc/systemd/system/
pi@raspberrypi:/etc/systemd/system $ ls
autologin@.service  dbus-org.freedesktop.timesync1.service  getty@tty1.service.d  poweroff.target.wants   sockets.target.wants wyzesense2mqtt.service  bluetooth.target.wants  default.target  graphical.target.wants       
rc-local.service.d  sshd.service dbus-fi.w1.wpa_supplicant1.service  dhcpcd5.service  halt.target.wants            reboot.target.wants     sysinit.target.wants 	dbus-org.bluez.service   display-manager.service                 
multi-user.target.wants      remote-fs.target.wants  syslog.service 	dbus-org.freedesktop.Avahi.service  getty.target.wants                      network-online.target.wants  samba-ad-dc.service     timers.target.wants
pi@raspberrypi:/etc/systemd/system $ nano wyzesense2mqtt.service

```
[... 81 lines cut ...]
   CGroup: /system.slice/wyzesense2mqtt.service
           `-13646 /bin/sleep 10

Aug 11 20:13:36 raspberrypi systemd[1]: Starting wyzesense2mqtt...

```

16) So I check the status again, and now I get a failure:
```
pi@raspberrypi:~ $ sudo systemctl status wyzesense2mqtt
* wyzesense2mqtt.service - wyzesense2mqtt
   Loaded: loaded (/etc/systemd/system/wyzesense2mqtt.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Tue 2020-08-11 20:33:59 EDT; 6ms ago
  Process: 29232 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 29362 ExecStart=/usr/bin/python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py (code=exited, status=1/FAILURE)
 Main PID: 29362 (code=exited, status=1/FAILURE)

```

17) Conclusion/wild guess: This is the line that doesn't make sense to me:
`Process: 29362 ExecStart=/usr/bin/python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py (code=exited, status=1/FAILURE)
`

In my wyzesense2mqtt.service file, the ExecStart line is this - so why is it different in the CLI error? Could this be my issue?
`ExecStart=python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py`

dale3h, 2020-08-18:
@ha14937 It looks like you've done everything 100% the same as me, so I am not sure which error is kicking back. Can you open a separate SSH session to the server and run this to view the log output while starting the service?

```shell
sudo journalctl -fu wyzesense2mqtt
```

This will open the logger journal and follow all log messages moving forward. Here's a breakdown of the command:

- `sudo` - Run the command as **s**uper**u**ser (root).
- `journalctl` - Query the systemd journal.
- `-f` - Show only the most recent journal entries, and continuously print new entries as they are appended to the journal.
- `-u wyzesense2mqtt` - Show messages for the specified systemd unit UNIT (such as a service unit).

(The manpage for `journalctl` can be found [here](https://manpages.debian.org/stretch/systemd/journalctl.1.en.html).)

The following line is just saying that the `/opt/wyzesense2mqtt/wyzesense2mqtt.py` script failed, but it doesn't give the specifics of why/how. That information will most likely be revealed when running the `sudo journalctl -fu wyzesense2mqtt` mentioned above.

```
Process: 29362 ExecStart=/usr/bin/python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py (code=exited, status=1/FAILURE)
```

dale3h, 2020-08-27:
@ha14937 Were you ever able to get this up and running?

ha14937, 2020-08-31:
@dale3h Hey Dale - thanks for following up. I've been on vacation so have not messed with it for a bit. I have time today and plan on giving it another shot. Will report back!

ha14937, 2020-08-31:
ok @dale3h here is what I have done:

To ensure I was starting from scratch, I went back and deleted ALL wyzesense2mqtt dirs/files from this machine,  rebooted, and went through each step I outlined above. When I check the status, I get:

```
pi@raspberrypi:/opt/wyzesense2mqtt/config $ sudo systemctl start wyzesense2mqtt
pi@raspberrypi:/opt/wyzesense2mqtt/config $ sudo systemctl status wyzesense2mqtt
* wyzesense2mqtt.service - wyzesense2mqtt
   Loaded: loaded (/etc/systemd/system/wyzesense2mqtt.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2020-08-31 09:19:15 EDT; 2s ago
  Process: 14195 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 14372 ExecStart=/usr/bin/python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py (code=exited, status=1/FAILURE)
 Main PID: 14372 (code=exited, status=1/FAILURE)

```

Then, per your advice, I opened a 2nd SSH session and executed the  `sudo journalctl -fu wyzesense2mqtt` command. Below are the results, though I'm not smart enough to understand what this log is telling me.

```
pi@raspberrypi:~ $ sudo journalctl -fu wyzesense2mqtt
-- Logs begin at Thu 2020-08-27 20:48:01 EDT. --
Aug 31 09:11:17 raspberrypi systemd[1]: wyzesense2mqtt.service: Scheduled restar                                                                                                             t job, restart counter is at 1.
Aug 31 09:11:17 raspberrypi systemd[1]: Stopped wyzesense2mqtt.
Aug 31 09:11:17 raspberrypi systemd[1]: Starting wyzesense2mqtt...
Aug 31 09:11:27 raspberrypi systemd[1]: Started wyzesense2mqtt.
Aug 31 09:11:27 raspberrypi python3[8130]: Traceback (most recent call last):
Aug 31 09:11:27 raspberrypi python3[8130]:   File "/opt/wyzesense2mqtt/wyzesense                                                                                                             2mqtt.py", line 15, in <module>
Aug 31 09:11:27 raspberrypi python3[8130]:     from retrying import retry
Aug 31 09:11:27 raspberrypi python3[8130]: ModuleNotFoundError: No module named                                                                                                              'retrying'
Aug 31 09:11:27 raspberrypi systemd[1]: wyzesense2mqtt.service: Main process exi                                                                                                             ted, code=exited, status=1/FAILURE
Aug 31 09:11:27 raspberrypi systemd[1]: wyzesense2mqtt.service: Failed with resu                                                                                                             lt 'exit-code'.
Aug 31 09:11:30 raspberrypi systemd[1]: wyzesense2mqtt.service: Service RestartSec=3s expired, scheduling restart.
Aug 31 09:11:30 raspberrypi systemd[1]: wyzesense2mqtt.service: Scheduled restart job, restart counter is at 2.
Aug 31 09:11:30 raspberrypi systemd[1]: Stopped wyzesense2mqtt.
Aug 31 09:11:30 raspberrypi systemd[1]: Starting wyzesense2mqtt...
Aug 31 09:11:40 raspberrypi systemd[1]: Started wyzesense2mqtt.
Aug 31 09:11:41 raspberrypi python3[8303]: Traceback (most recent call last):
Aug 31 09:11:41 raspberrypi python3[8303]:   File "/opt/wyzesense2mqtt/wyzesense2mqtt.py", line 15, in <module>
Aug 31 09:11:41 raspberrypi python3[8303]:     from retrying import retry
Aug 31 09:11:41 raspberrypi python3[8303]: ModuleNotFoundError: No module named 'retrying'
Aug 31 09:11:41 raspberrypi systemd[1]: wyzesense2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Aug 31 09:11:41 raspberrypi systemd[1]: wyzesense2mqtt.service: Failed with result 'exit-code'.
Aug 31 09:11:44 raspberrypi systemd[1]: wyzesense2mqtt.service: Service RestartSec=3s expired, scheduling restart.
Aug 31 09:11:44 raspberrypi systemd[1]: wyzesense2mqtt.service: Scheduled restart job, restart counter is at 3.
Aug 31 09:11:44 raspberrypi systemd[1]: Stopped wyzesense2mqtt.
Aug 31 09:11:44 raspberrypi systemd[1]: Starting wyzesense2mqtt...
Aug 31 09:11:54 raspberrypi systemd[1]: Started wyzesense2mqtt.
Aug 31 09:11:55 raspberrypi python3[8487]: Traceback (most recent call last):
Aug 31 09:11:55 raspberrypi python3[8487]:   File "/opt/wyzesense2mqtt/wyzesense2mqtt.py", line 15, in <module>
Aug 31 09:11:55 raspberrypi python3[8487]:     from retrying import retry
Aug 31 09:11:55 raspberrypi python3[8487]: ModuleNotFoundError: No module named 'retrying'
Aug 31 09:11:55 raspberrypi systemd[1]: wyzesense2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Aug 31 09:11:55 raspberrypi systemd[1]: wyzesense2mqtt.service: Failed with result 'exit-code'.
Aug 31 09:11:58 raspberrypi systemd[1]: wyzesense2mqtt.service: Service RestartSec=3s expired, scheduling restart.
Aug 31 09:11:58 raspberrypi systemd[1]: wyzesense2mqtt.service: Scheduled restart job, restart counter is at 4.
Aug 31 09:11:58 raspberrypi systemd[1]: Stopped wyzesense2mqtt.
Aug 31 09:11:58 raspberrypi systemd[1]: Starting wyzesense2mqtt...
Aug 31 09:12:08 raspberrypi systemd[1]: Started wyzesense2mqtt.
Aug 31 09:12:09 raspberrypi python3[8707]: Traceback (most recent call last):
Aug 31 09:12:09 raspberrypi python3[8707]:   File "/opt/wyzesense2mqtt/wyzesense2mqtt.py", line 15, in <module>
Aug 31 09:12:09 raspberrypi python3[8707]:     from retrying import retry
Aug 31 09:12:09 raspberrypi python3[8707]: ModuleNotFoundError: No module named 'retrying'
```

dale3h, 2020-09-03:
@clandestine-avocado The logs are saying that the Python module `retrying` is missing. Did you install the dependencies when you started over? This is listed as step 7 under [**Linux Systemd**](https://github.com/raetha/wyzesense2mqtt/tree/6d3bf5d14a643762403637a1b8d745b98422366d#linux-systemd) on the README page.

> 7. Install dependencies
> 
> ```
> pip3 install -r requirements.txt
> ```

clandestine-avocado, 2020-09-04:
@dale3h - I wasn't sure, so I just went back and pip'd the requirements.txt - and it looks like all the reqs were there already.

```
pi@raspberrypi:/opt/wyzesense2mqtt $ pip3 install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: paho_mqtt in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 5)) (1.5.0)
Requirement already satisfied: retrying in /home/pi/.local/lib/python3.7/site-packages (from -r requirements.txt (line 6)) (1.3.3)
Requirement already satisfied: pyyaml in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 7)) (5.3.1)
Requirement already satisfied: docopt in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 12)) (0.6.2)
Requirement already satisfied: six>=1.7.0 in /usr/lib/python3/dist-packages (from retrying->-r requirements.txt (line 6)) (1.12.0)

```

dale3h, 2020-09-04:
@clandestine-avocado It looks like you've installed the requirements to the `pi` user's local python installation, which means `root` will not be able to see that. Can you try this and see if it fixes the issue?

```
$ cd /opt/wyzesense2mqtt
$ pip3 install -r requirements.txt
```

This will install the requirements into the python that is available to the root user.

clandestine-avocado, 2020-09-04:
You're the man - that did it - **no errors!** 

It added the one test sensor I placed in my wyzesense2mqtt config to HA - however it's not reacting to motion (its a PIR).

dale3h, 2020-09-04:
Awesome! I am really glad to hear that it finally launched without any errors!

Can you try removing it and re-pairing it to see if that does the trick? It shouldn't need to be paired again, but sometimes things get a little quirky.

clandestine-avocado, 2020-09-04:
@dale3h - I'm up and running! I had to launch the bridge tool using sudo (`sudo python3 bridge_tool_cli.py --device /dev/hidraw0`) and re-pair all my devices. A small price to pay to get this up and running again. 

Thanks again for your time. 

I think, when I get some time, I will try to help update the documentation for the Systemd method.

RandomPotato893, 2020-10-31:
I am having the same problem and I tried all the steps above.  this is my log output. 

Oct 31 03:12:20 ubuntuserv systemd[1]: Starting wyzesense2mqtt...
Oct 31 03:12:30 ubuntuserv systemd[1]: Started wyzesense2mqtt.
Oct 31 03:12:31 ubuntuserv python3[1564]: Logging initialized...
Oct 31 03:12:31 ubuntuserv python3[1564]: Reading configuration...
Oct 31 03:12:31 ubuntuserv python3[1564]: Connecting to MQTT host 192.168.1.3
Oct 31 03:12:31 ubuntuserv python3[1564]: Connecting to dongle auto
Oct 31 03:12:31 ubuntuserv python3[1564]: No device found on path auto
Oct 31 03:12:31 ubuntuserv python3[1564]: Reading sensors configuration...
Oct 31 03:12:31 ubuntuserv python3[1564]: Traceback (most recent call last):
Oct 31 03:12:31 ubuntuserv python3[1564]:   File "/opt/wyzesense2mqtt/wyzesense2mqtt.py", line 449, in <module>
Oct 31 03:12:31 ubuntuserv python3[1564]:     init_sensors()
Oct 31 03:12:31 ubuntuserv python3[1564]:   File "/opt/wyzesense2mqtt/wyzesense2mqtt.py", line 154, in init_sensors
Oct 31 03:12:31 ubuntuserv python3[1564]:     result = WYZESENSE_DONGLE.List()
Oct 31 03:12:31 ubuntuserv python3[1564]: NameError: name 'WYZESENSE_DONGLE' is not defined
Oct 31 03:12:31 ubuntuserv systemd[1]: wyzesense2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Oct 31 03:12:31 ubuntuserv systemd[1]: wyzesense2mqtt.service: Failed with result 'exit-code'.
Oct 31 03:12:34 ubuntuserv systemd[1]: wyzesense2mqtt.service: Scheduled restart job, restart counter is at 3.
Oct 31 03:12:34 ubuntuserv systemd[1]: Stopped wyzesense2mqtt.

Any help is appreciated.

raetha, 2020-11-28:
@clandestine-avocado, @jeffrutland , @dale3h the devel branch now has fixes for much of what you all chatted about. It will now start with missing sensors.yaml and logging.yaml files. It will also automatically copy the sample config.yaml file, but things won't work until at least the hostname is added to that. I also updated the readme with all the notes I could find about missing steps or bad commands. Please take a look, and maybe test out the devel branch some to help make sure it's good. Hoping to move the changes to the master branch this weekend.

@RandomPotato893 in your case the logs show that it can't find your bridge device. Can you confirm /dev/hidraw0 exists and that the bridge device is plugged in and detected by the kernel? The auto setting also may not work if you'd create a udev rule to change the device name. In that case, please hard code your device path into the config.yaml file. If neither of those ideas solve help get you going, can you create a new to track this and we'll help out as we can?

raetha, 2025-03-03:
Closing based on last comment being that most of this was incorporated into updates back in 2020 with no further issues reported after that.

closed 2025-03-03 by raetha