raetha/wyzesense2mqtt #42: Socket error on client wyzesense2mqtt, disconnecting

SamuelCarson, 2020-09-14:
I had wyzesense2mqtt working on bare metal raspbian. For various reasons I need to switch to docker. 
Installed Products
Docker version 19.03.12, build 48a6621
docker-compose version 1.27.2, build unknown
and latest raetha/wyzesense2mqtt

Container starts and attempts to connnect to mqtt.
Mqtt logs show - (https://github.com/raetha/wyzesense2mqtt/files/5220180/wyzesense2mqtt-mqtt.log)
wyzesense2mqtt.logs show
(https://github.com/raetha/wyzesense2mqtt/files/5220179/wyzesense2mqtt.log)

Mqtt credentials verified.

I don't understand the problem.

AK5nowman, 2020-09-28:
Your wyzesense2mqtt.log shows you that it isn't finding the USB device. I don't really know much about docker but are you attaching the dongle to it correctly?

raetha, 2020-11-28:
@SamuelCarson, did you make sure to pass the /dev/hidraw0 device to the Docker container? I think my example shows how to do it in Compose. Also, if you used udev to rename the Wyze dongle device, you not only need to pass the renamed device to Docker, but also will likely need to hard code the value in the config.yaml file. I did some testing with friendly udev names and it never went well unless I hard coded everything.