raetha/wyzesense2mqtt #43: No MQTT messages

PeterSelvaraj, 2020-09-20:
Raetha and creators: Thank you for creating this software and all your hard work.

## Describe the Bug

I took the docker route and got things up and running pretty quickly on a raspberry pi 4.

Here are a few things that differentiate my experience from what is described in the instructions. The instructions say that the sensors.yaml file would be created. That did not happen. I was expecting, perhaps it would be created when the pairing is completed. But that did not happen. After spending some time trying to get it to work, I decided to create the file and see if that would help. The moved things forward to a different set of initialization errors. I decided to add the sample sensors to see if that would help. That moved things forward, the new sensors were added. After which I removed the sample sensors. Things moved smoothly this point forward and the devices and sensors appeared in MQTT messages and were added to corresponding homeassistant lists.

However, I could not move forward beyond this point. I did not see any MQTT messages after the initial discovery messages. The entries in home assistant for battery level and signal strength never updated and continue to display unknown.

I've looked at docker log and wyze log. There are no errors displayed. Following are the contents of the relevant files.

config.yaml

mqtt_host: 192.168.10.3
mqtt_port: 1883
mqtt_username: hass
mqtt_password: ****
mqtt_client_id: wyzesense2mqtt
mqtt_clean_session: false
mqtt_keepalive: 60
mqtt_qos: 2
mqtt_retain: true
self_topic_root: wyzesense2mqtt
hass_topic_root: homeassistant
publish_sensor_name: true
usb_dongle: auto

sensors.yaml

778337E6:
  class: motion
  invert_state: false
  name: Wyze Sense 778337E6
77933B2C:
  class: door
  invert_state: false
  name: Wyze Sense 77933B2C

logging.yaml

version: 1
formatters:
  simple:
    format: '%(message)s'
  verbose:
    datefmt: '%Y-%m-%d %H:%M:%S'
    format: '%(asctime)s %(levelname)-8s %(name)-15s %(message)s'
handlers:
  console:
    class: logging.StreamHandler
    formatter: simple
    level: DEBUG
  file:
    backupCount: 7
    class: logging.handlers.TimedRotatingFileHandler
    encoding: utf-8
    filename: logs/wyze.log
    formatter: verbose
    level: INFO
    when: midnight
root:
  handlers:
    - file
    - console
  level: DEBUG

wyze.log

2020-09-20 13:35:02 INFO     wyzesense2mqtt  Connecting to MQTT host 192.168.10.3
2020-09-20 13:35:02 INFO     wyzesense2mqtt  Connecting to dongle /dev/hidraw0
2020-09-20 13:35:03 INFO     wyzesense2mqtt  Publishing discovery topics for 778337E6
2020-09-20 13:35:03 INFO     wyzesense2mqtt  Publishing discovery topics for 77933B2C
2020-09-20 13:35:03 INFO     wyzesense2mqtt  Connected to MQTT: return code 0
2020-09-20 13:40:09 INFO     wyzesense2mqtt  Disconnected from MQTT: return code 1
2020-09-20 13:41:32 INFO     wyzesense2mqtt  Connecting to MQTT host 192.168.10.3
2020-09-20 13:41:33 INFO     wyzesense2mqtt  Connecting to dongle /dev/hidraw0
2020-09-20 13:41:34 INFO     wyzesense2mqtt  Publishing discovery topics for 778337E6
2020-09-20 13:41:34 INFO     wyzesense2mqtt  Publishing discovery topics for 77933B2C
2020-09-20 13:41:34 INFO     wyzesense2mqtt  Connected to MQTT: return code 0
2020-09-20 15:31:02 INFO     wyzesense2mqtt  Connecting to MQTT host 192.168.10.3
2020-09-20 15:31:02 INFO     wyzesense2mqtt  Connecting to dongle /dev/hidraw0
2020-09-20 15:31:03 INFO     wyzesense2mqtt  Publishing discovery topics for 778337E6
2020-09-20 15:31:03 INFO     wyzesense2mqtt  Publishing discovery topics for 77933B2C
2020-09-20 15:31:03 INFO     wyzesense2mqtt  Connected to MQTT: return code 0
2020-09-20 15:51:27 INFO     wyzesense2mqtt  In on_message_reload: 
2020-09-20 15:51:27 INFO     wyzesense2mqtt  Publishing discovery topics for 778337E6
2020-09-20 15:51:27 INFO     wyzesense2mqtt  Publishing discovery topics for 77933B2C

docker log
[docker.log](https://github.com/raetha/wyzesense2mqtt/files/5252661/docker.log)

I've tried reloading the sensors by publishing the reload message. I've tried restarting the docker container, rebooting the pi. But I've not been able to move beyond the sensors and devices appearing in home assistant.

I'm assuming that any time the sensor changes, mqtt messages will be published. Forgive me, I'm new the automation world and am slowly getting up to speed on things. The issue appears to be lack of broadcast of mqtt messages when things change. I've no idea what I can to solve this/move forward. I'd greatly appreciate any help you can provide. Thank you for taking the time to read this long post.

AK5nowman, 2020-09-22:
What are you using to see the mqtt messages being published to wyzesense2mqtt? In the logging.yaml file, if you change the file logging level to debug you will be able to see the payload that should be published. That will at least tell you if its getting that far.

PeterSelvaraj, 2020-09-22:
> What are you using to see the mqtt messages being published to wyzesense2mqtt? In the logging.yaml file, if you change the file logging level to debug you will be able to see the payload that should be published. That will at least tell you if its getting that far.

I'm using homeassistant MQTT integration to see the messages coming through by subscribing to "#" topic which results in all messages being captured. But, what you're suggesting is a better approach, check things out right at the source. I will look into it and report back on what I find. Thank you for your reply.

PeterSelvaraj, 2020-09-22:
> What are you using to see the mqtt messages being published to wyzesense2mqtt? In the logging.yaml file, if you change the file logging level to debug you will be able to see the payload that should be published. That will at least tell you if its getting that far.

I've attached the log after updating logging preference to debug. I see that the initial discovery messages are sent. After that interactions with the sensors don't seem to trigger any new messages. Not sure why that is the case. I don't see any error messages in the log unless I'm overlooking something.
[wyze.log](https://github.com/raetha/wyzesense2mqtt/files/5264141/wyze.log)

PeterSelvaraj, 2020-09-22:
> What are you using to see the mqtt messages being published to wyzesense2mqtt? In the logging.yaml file, if you change the file logging level to debug you will be able to see the payload that should be published. That will at least tell you if its getting that far.

I was using homeassistant to listen to all events coming across by subscribing to "#" and reviewing all data logged. I saw initial discovery messages but did not see any other messages but your approach will certainly made it easier. I changed the logging to debug level and noticed the same thing. The discovery messages were being sent but further interactions did not produce any messages. I'm attaching the log for review.
[wyze.log](https://github.com/raetha/wyzesense2mqtt/files/5264533/wyze.log)

AK5nowman, 2020-09-23:
Huh, interesting. It is like its not receiving events from the dongle. Can you try running the bridge_tool_cli.py utility and toggle your sensors and see if any events print to the screen. Nothing in your logs indicate any sort of failure (not to say it isn't failing silently somewhere)

PeterSelvaraj, 2020-09-25:
I used the utility and it showed that the devices were paired. I thought perhaps removing the devices and pairing again with the utility might solve the problem. The removal went smoothly. However, when I attempted to pair the devices, errors were displayed unfortunately I didn't record what they were, but it alluded to missing sensors.yaml file. So I created a copy of the sample sensors.yaml file and that fixed the errors. All devices were paired successfully and the mqtt messages were being sent as expected. Last time when I paired I noticed that the blue light wasn't blinking. This time however, when I was using the utility, the blue light was blinking during pairing and stopped blinking when the pairing was complete.

Thanks AK5nowman for your pointers and helping me getting things working. Much appreciated!

closed 2020-11-28 by raetha