raetha/wyzesense2mqtt #44: Sensors.yaml file not optional

AK5nowman, 2020-09-22:
## Describe the Bug

the Sensors.yaml file has to be defined and it has to had at least a single entry, contrary to what the instructions set says. There are no checks to see if the SENSORS global variable is set throughout the code and will fail silently in multiple locations. 

### Steps To Reproduce

Steps to reproduce the behavior:

1. Remove sensors.yaml file and start up

### Expected Behavior

I expect the app to create and write the sensor data to the sensor.yaml file

tehstevo, 2020-11-25:
Had the same experience in my install tonight.  Putting a bogus entry into the sensors.yaml file resolved it.

raetha, 2020-11-28:
This is now fixed in the devel branch. Thanks @AK5nowman for the PR that gave an easy way to handle this.

Will go to master branch with 1.2, hopefully this weekend.

commit 0f710f19 (2020-11-28): Handle not having a sensors.yaml file on startup (Fixes #44, Reference PR #45)
Logging enhancements (Reference PR #45)
Add support to disable HASS discovery topics (Reference PR #45)
Refactored configuration initialization to be self-updating
Removing a sensor now removes it from config file (Reference PR #45)
PR #41 (2021-02-11): 1.2
Version bump to 1.2
Update bug report template to use markdown (@dale3h)
Resolve most flake8 issues (@dale3h)
Bridge connection retries should now continue when an IOError is triggered (#17)
Add vim to docker image (Resolves #39)
Add Neos Leak Sensor support (Resolves #29) - In Progress
Handle not having a sensors.yaml file on startup (Fixes #44, Reference PR #45 @AK5nowman)
Logging enhancements (Reference PR #45 @AK5nowman)
Add support to disable HASS discovery topics (Reference PR #45 @AK5nowman)
Refactored configuration initialization to be self-updating for new features
Removing a sens [... cut]
closed 2021-02-11 by raetha