raetha/wyzesense2mqtt #46: MQTT dropped off 3 days ago - but CLI prints motion events.

clandestine-avocado, 2020-09-27:
## Describe the Bug

A clear and concise description of what the bug is.
Almost identical to #43  except A) I'm running it via Systemd and B) I've had this up and running for weeks with no issues - and it just stopped sending new MQTT messages.

### Steps To Reproduce

Steps to reproduce the behavior:
1. Start the service, and check it's status - all running normally with no errors.
2. Launch the CLI tool, L to list the sensors to verify all 5 are still paired - they are.
3. Manually trip sensors - JSON appears under the CLI tool.
4. Checked the log, looks exactly like the CLI output
5. --- At this point, I'm pretty sure it IS sending a MQTT message----
6. So I head over to HA to see if I can subscribe/listen via topic `wyzesense2mqtt/<MAC of sensor>`
7. When I hit listen in HA, it prints the last msg it received (from ~ 3 days ago)
[image]
8. I trip the sensor, and no new msgs appear.
9. Double check topic subscription in device more info menu. Topic is correct `wyzesense2mqtt/<MAC of sensor>`
[image]

9. I checked the Mosquitto broker log - and the client is connecting and disconnecting normally.
10. Ran out of troubleshooting ideas and came here looking for assistance. :)

### Expected Behavior

Back to working smoothly as it has for weeks

### Screenshots

See above - can add more if needed.

### Desktop

Please complete the following information:

- OS: Raspbian
- Browser: Headless
- Version: up to date

### Smartphone

N/A
## Additional Context

Add any other context about the problem here.

AK5nowman, 2020-09-27:
Can you post the logs from the logging directory? I don't think this is related to to #43 because that issue he wasn't seeing any events at all, even with the CLI tool. 

If you restart the service does it begin to work again? If not, please change file logging level to debug and restart + trigger the devices. With a logging level of debug it will print is message payload to the file as it would publish it to the broker.

clandestine-avocado, 2020-09-27:
Sure thing! I have tried reloading, then restarting with the below commands.

```
pi@raspberrypi:/etc/wyzesense2mqtt/wyzesense2mqtt $ sudo systemctl stop wyzesense2mqtt
pi@raspberrypi:/etc/wyzesense2mqtt/wyzesense2mqtt $ sudo systemctl daemon-reload
pi@raspberrypi:/etc/wyzesense2mqtt/wyzesense2mqtt $ sudo systemctl start wyzesense2mqtt
pi@raspberrypi:/etc/wyzesense2mqtt/wyzesense2mqtt $ sudo systemctl status wyzesense2mqtt
* wyzesense2mqtt.service - wyzesense2mqtt
   Loaded: loaded (/etc/systemd/system/wyzesense2mqtt.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-09-26 20:35:44 EDT; 15s ago
  Process: 6253 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
 Main PID: 6415 (python3)
    Tasks: 2 (limit: 2068)
   CGroup: /system.slice/wyzesense2mqtt.service
           `-6415 /usr/bin/python3 /opt/wyzesense2mqtt/wyzesense2mqtt.py

Sep 26 20:35:46 raspberrypi python3[6415]:   {"name": "Kitchen Zone 2 Wyze 779B20D8 Battery", "dev_cla": "battery", "unit_of_meas": "%", "val_tpl": "{{ value_json.battery }}", "uniq_id": "wy
Sep 26 20:35:46 raspberrypi python3[6415]: sensor_mac: 779B23A7
Sep 26 20:35:46 raspberrypi python3[6415]: Publishing discovery topics for 779B23A7
Sep 26 20:35:46 raspberrypi python3[6415]:   homeassistant/binary_sensor/wyzesense_779B23A7/state/config
Sep 26 20:35:46 raspberrypi python3[6415]:   {"name": "Living Room Wyze 779B23A7", "dev_cla": "motion", "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2mqtt/779B23A7", "val_tpl": "{{ 
Sep 26 20:35:46 raspberrypi python3[6415]:   homeassistant/sensor/wyzesense_779B23A7/signal_strength/config
Sep 26 20:35:46 raspberrypi python3[6415]:   {"name": "Living Room Wyze 779B23A7 Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_stren
Sep 26 20:35:46 raspberrypi python3[6415]:   homeassistant/sensor/wyzesense_779B23A7/battery/config
Sep 26 20:35:46 raspberrypi python3[6415]:   {"name": "Living Room Wyze 779B23A7 Battery", "dev_cla": "battery", "unit_of_meas": "%", "val_tpl": "{{ value_json.battery }}", "uniq_id": "wyzes
Sep 26 20:35:46 raspberrypi python3[6415]: Connected to MQTT: return code 0

```

**Logs:**
It was already in debug mode - sorry about screen shot rather than text - it won't let me copy out of the log file for some reason. Happy to post the whole thing if I can somehow figure out how to allow copy/paste. Sensors look active, and the MQTT return code of 0 means successful connection I believe.

[image]

While doing the stop/start/status dance, I "manually" subscribed/listened to topic `wyzesense2mqtt/779B23A7` to see if what the logs say is correct - specifically the "Publishing the discovery topics for 779B23A7" - and nothing come thru into HA. I DO get the last message from a few days ago when I click "listen" in HA, but that's it (this is 1st  screen shot in my OP)

AK5nowman, 2020-09-27:
Are you sure you are in debug mode for file logging? There are two sections in the logging.yaml file that you can set the debug level. I ask because I don’t see any debug messages and if you are truly in debug then it’s failing silently in about 9 line of code. Can you post your sensor.yaml contents as well as your config.yaml contents (remove your credentials if you care)

clandestine-avocado, 2020-09-27:
I was not in DEBUG - thank you. I did not realize there were _two_ log level settings; switched it to DEBUG now, logs at the bottom. To clarify, I am looking at log files at cd /etc/wyzesense2mqtt/wyzesense2mqtt/logs - which I believe is correct. 

```
pi@raspberrypi:~/wyzesense2mqtt/wyzesense2mqtt/config $ cd /etc/wyzesense2mqtt/wyzesense2mqtt/logs 
pi@raspberrypi:/etc/wyzesense2mqtt/wyzesense2mqtt/logs $ ls -l
total 584
-rw-r--r-- 1 root root  28730 Sep 27 14:21 wyzesense2mqtt.log
-rw-r--r-- 1 root root 103644 Sep 20 23:10 wyzesense2mqtt.log.2020-09-20
-rw-r--r-- 1 root root  69438 Sep 21 23:57 wyzesense2mqtt.log.2020-09-21
-rw-r--r-- 1 root root  66351 Sep 22 22:52 wyzesense2mqtt.log.2020-09-22
-rw-r--r-- 1 root root  69916 Sep 23 22:28 wyzesense2mqtt.log.2020-09-23
-rw-r--r-- 1 root root  59492 Sep 24 23:31 wyzesense2mqtt.log.2020-09-24
-rw-r--r-- 1 root root  30150 Sep 25 23:55 wyzesense2mqtt.log.2020-09-25
-rw-r--r-- 1 root root 115284 Sep 26 23:46 wyzesense2mqtt.log.2020-09-26
```

**Here is my config:**
```
mqtt_port: 1883
mqtt_username: ********
mqtt_password: ********************
mqtt_client_id: wyzesense2mqtt
mqtt_clean_session: false
mqtt_keepalive: 60
mqtt_qos: 0
mqtt_retain: true
self_topic_root: wyzesense2mqtt
hass_topic_root: homeassistant
publish_sensor_name: true
usb_dongle: auto
```
**Here is my logging.yaml:**
```
version: 1
formatters:
  simple:
    format: '%(message)s'
  verbose:
    datefmt: '%Y-%m-%d %H:%M:%S'
    format: '%(asctime)s %(levelname)-8s %(name)-15s %(message)s'
handlers:
  console:
    class: logging.StreamHandler
    formatter: simple
    level: DEBUG
  file:
    backupCount: 7
    class: logging.handlers.TimedRotatingFileHandler
    encoding: utf-8
    filename: logs/wyzesense2mqtt.log
    formatter: verbose
    level: DEBUG # This was set to INFO previously, just changed to DEBUG and restarted.
    when: midnight
root:
  handlers:
    - file
    - console
  level: DEBUG
```

**Here are the DEBUG logs:**
```
2020-09-27 14:35:33 DEBUG    wyzesense2mqtt  Logging initialized...
2020-09-27 14:35:33 DEBUG    wyzesense2mqtt  Reading configuration...
2020-09-27 14:35:33 INFO     wyzesense2mqtt  Connecting to MQTT host 192.168.1.237
2020-09-27 14:35:33 INFO     wyzesense2mqtt  Connecting to dongle /dev/hidraw0
2020-09-27 14:35:34 DEBUG    wyzesense2mqtt    MAC: 77863C6F,  VER: 0.0.0.30 V1.4 Dongle UD3U,  ENR: b'\x9b]\x03\xefHU\xb6\xf7s\xf0\xb2\xe6i\xbd/\x95'
2020-09-27 14:35:34 DEBUG    wyzesense2mqtt  Reading sensors configuration...
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt  Linked sensors: ['779B20D8', '779B1EC8', '778CA525', '778CA535', '779B23A7']
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt  sensor_mac: 779B20D8
[... 30 lines cut ...]
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    homeassistant/binary_sensor/wyzesense_779B20D8/state/config
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    {"name": "Kitchen Zone 2 Wyze 779B20D8", "dev_cla": "motion", "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2mqtt/779B20D8", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_779B20D8_state", "stat_t": "wyzesense2mqtt/779B20D8", "dev": {"identifiers": ["wyzesense_779B20D8", "779B20D8"], "manufacturer": "Wyze", "model": "Sense Motion Sensor", "name": "Kitchen Zone 2 Wyze 779B20D8", "sw_version": ""}}
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    homeassistant/sensor/wyzesense_779B20D8/signal_strength/config
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    {"name": "Kitchen Zone 2 Wyze 779B20D8 Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_strength }}", "uniq_id": "wyzesense_779B20D8_signal_strength", "stat_t": "wyzesense2mqtt/779B20D8", "dev": {"identifiers": ["wyzesense_779B20D8", "779B20D8"], "manufacturer": "Wyze", "model": "Sense Motion Sensor", "name": "Kitchen Zone 2 Wyze 779B20D8", "sw_version": ""}}
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    homeassistant/sensor/wyzesense_779B20D8/battery/config
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    {"name": "Kitchen Zone 2 Wyze 779B20D8 Battery", "dev_cla": "battery", "unit_of_meas": "%", "val_tpl": "{{ value_json.battery }}", "uniq_id": "wyzesense_779B20D8_battery", "stat_t": "wyzesense2mqtt/779B20D8", "dev": {"identifiers": ["wyzesense_779B20D8", "779B20D8"], "manufacturer": "Wyze", "model": "Sense Motion Sensor", "name": "Kitchen Zone 2 Wyze 779B20D8", "sw_version": ""}}
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt  sensor_mac: 779B23A7
2020-09-27 14:35:35 INFO     wyzesense2mqtt  Publishing discovery topics for 779B23A7
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    homeassistant/binary_sensor/wyzesense_779B23A7/state/config
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    {"name": "Living Room Wyze 779B23A7", "dev_cla": "motion", "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2mqtt/779B23A7", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_779B23A7_state", "stat_t": "wyzesense2mqtt/779B23A7", "dev": {"identifiers": ["wyzesense_779B23A7", "779B23A7"], "manufacturer": "Wyze", "model": "Sense Motion Sensor", "name": "Living Room Wyze 779B23A7", "sw_version": ""}}
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    homeassistant/sensor/wyzesense_779B23A7/signal_strength/config
2020-09-27 14:35:35 DEBUG    wyzesense2mqtt    {"name": "Living Room Wyze 779B23A7 Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_strength }}", "uniq_id": "wyzesense_779B23A7_signal_strength", "stat_t": "wyzese [... cut ...]
2020-09-27 14:37:31 DEBUG    wyzesense2mqtt  {'available': True, 'mac': '779B23A7', 'device_class': 'motion', 'last_seen': 1601231848.655, 'last_seen_iso': '2020-09-27T14:37:28.655000', 'signal_strength': -61, 'battery': 94, 'name': 'Living Room Wyze 779B23A7', 'state': 0}
2020-09-27 14:38:20 DEBUG    wyzesense2mqtt  sensor_mac: 779B23A7
2020-09-27 14:38:20 DEBUG    wyzesense2mqtt  Non-state event data: [2020-09-27 14:38:15][779B23A7]RawEvent: type=raw_AB, data=b'020200020100000000020000003d00'

```

AK5nowman, 2020-09-27:
Hmm okay. Yeah the two 'handlers', one for when you are watching logs via the console and one for the messages you want to write to disk. 

` {'available': True, 'mac': '779B23A7', 'device_class': 'motion', 'last_seen': 1601231848.655, 'last_seen_iso': '2020-09-27T14:37:28.655000', 'signal_strength': -61, 'battery': 94, 'name': 'Living Room Wyze 779B23A7', 'state': 0}`

So this is the MQTT message that it is attempting to publish. Either the message doesn't get delivered or it fails to send. To test out the first one lets change your mqtt_qos to either 1 or 2, so that it is guaranteed to get to the broker. In the mean time I'm going to play around with trying to capture some errors on the publish call.

clandestine-avocado, 2020-09-28:
OK, mqtt_qos is now set to 2, and I made the edits to wyzesense2mqtt.py. 

Below are the logs after restart - the same logs [here ](https://hastebin.com/niroceyuva.php) if they are easier for you to read outside of GH. Device being trigger by me for testing is MAC 779B23A7. I don't really understand everything in the log - but it looks like its trying to send it over and over. But is also looks like its being received, based on the lines that say `Received PINGRESP`

```
2020-09-27 20:34:04 DEBUG    wyzesense2mqtt  Logging initialized...
2020-09-27 20:34:04 DEBUG    wyzesense2mqtt  Reading configuration...
2020-09-27 20:34:04 INFO     wyzesense2mqtt  Connecting to MQTT host 192.168.1.237
2020-09-27 20:34:04 DEBUG    wyzesense2mqtt  Sending CONNECT (u1, p1, wr0, wq0, wf0, c0, k60) client_id=b'wyzesense2mqtt-1'
2020-09-27 20:34:04 INFO     wyzesense2mqtt  Connecting to dongle /dev/hidraw0
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    MAC: 77863C6F,  VER: 0.0.0.30 V1.4 Dongle UD3U,  ENR: b'\x9b]\x03\xefHU\xb6\xf7s\xf0\xb2\xe6i\xbd/\x95'
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Reading sensors configuration...
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Linked sensors: ['779B20D8', '779B1EC8', '778CA525', '778CA535', '779B23A7']
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 779B20D8
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 779B1EC8
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 778CA525
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 778CA535
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 779B23A7
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 778CA525
2020-09-27 20:34:05 INFO     wyzesense2mqtt  Publishing discovery topics for 778CA525
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Sending PUBLISH (d0, q2, r1, m1), 'b'homeassistant/binary_sensor/wyzesense_778CA525/state/config'', ... (413 bytes)
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    homeassistant/binary_sensor/wyzesense_778CA525/state/config
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    {"name": "wyzesense con washer lid", "dev_cla": "opening", "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2mqtt/778CA525", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_778CA525_state", "stat_t": "wyzesense2mqtt/778CA525", "dev": {"identifiers": ["wyzesense_778CA525", "778CA525"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "wyzesense con washer lid", "sw_version": ""}}
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Sending PUBLISH (d0, q2, r1, m2), 'b'homeassistant/sensor/wyzesense_778CA525/signal_strength/config'', ... (409 bytes)
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    homeassistant/sensor/wyzesense_778CA525/signal_strength/config
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    {"name": "wyzesense con washer lid Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_strength }}", "uniq_id": "wyzesense_778CA525_signal_strength", "stat_t": "wyzesense2mqtt/778CA525", "dev": {"identifiers": ["wyzesense_778CA525", "778CA525"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "wyzesense con washer lid", "sw_version": ""}}
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Sending PUBLISH (d0, q2, r1, m3), 'b'homeassistant/sensor/wyzesense_778CA525/battery/config'', ... (375 bytes)
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    homeassistant/sensor/wyzesense_778CA525/battery/config
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    {"name": "wyzesense con washer lid Battery", "dev_cla": "battery", "unit_of_meas": "%", "val_tpl": "{{ value_json.battery }}", "uniq_id": "wyzesense_778CA525_battery", "stat_t": "wyzesense2mqtt/778CA525", "dev": {"identifiers": ["wyzesense_778CA525", "778CA525"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "wyzesense con washer lid", "sw_version": ""}}
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  sensor_mac: 778CA535
2020-09-27 20:34:05 INFO     wyzesense2mqtt  Publishing discovery topics for 778CA535
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Sending PUBLISH (d0, q2, r1, m4), 'b'homeassistant/binary_sensor/wyzesense_778CA535/state/config'', ... (403 bytes)
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    homeassistant/binary_sensor/wyzesense_778CA535/state/config
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    {"name": "Wyze Sense 778CA535", "dev_cla": "opening", "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2mqtt/778CA535", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_778CA535_state", "stat_t": "wyzesense2mqtt/778CA535", "dev": {"identifiers": ["wyzesense_778CA535", "778CA535"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense 778CA535", "sw_version": ""}}
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt  Sending PUBLISH (d0, q2, r1, m5), 'b'homeassistant/sensor/wyzesense_778CA535/signal_strength/config'', ... (399 bytes)
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    homeassistant/sensor/wyzesense_778CA535/signal_strength/config
2020-09-27 20:34:05 DEBUG    wyzesense2mqtt    {"name": "Wyze Sense 778CA535 Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_strength }}", "uniq_id": "wyzesense_778CA535_signal_strength", "stat_t": "wyzesense2mqtt/778CA535", "dev": {"identifiers": ["wyzesense_778CA535", "778CA535"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense 778CA535", "sw_version": ""}}
[... cut ...]
2020-09-27 20:42:13 DEBUG    wyzesense2mqtt  Sending PUBLISH (d1, q2, r1, m20), 'b'wyzesense2mqtt/779B23A7'', ... (228 bytes)
2020-09-27 20:42:17 DEBUG    wyzesense2mqtt  Sending PUBLISH (d1, q2, r1, m17), 'b'wyzesense2mqtt/779B20D8'', ... (231 bytes)
2020-09-27 20:42:17 DEBUG    wyzesense2mqtt  Sending PUBLISH (d1, q2, r1, m19), 'b'wyzesense2mqtt/779B23A7'', ... (228 bytes)
2020-09-27 20:42:19 DEBUG    wyzesense2mqtt  Sending PUBLISH (d1, q2, r1, m21), 'b'wyzesense2mqtt/779B23A7'', ... (228 bytes)

```

AK5nowman, 2020-09-28:
Wait a second, you updated the client ID and restarted wyzesense2mqtt and it all started working for you? As in problem fixed?

If so, do you have any other clients writing to this same broker with the same client ID? Seems like HA shouldn't care (and doesn't know?) what the client ID of the publishing client is.

clandestine-avocado, 2020-09-28:
@AK5nowman yes - I am confused as you are. 

When I was in the `config.yaml` changing the `mqtt_qos` to 2, I noticed the line that defines the client ID. I then looked at the Mosquito add-on log to see if that ID (was defined as "wyzesense2mqtt-1") showed up in the log. It did not, and I thought the "-1" on the end of the current clientID was odd - I don't recall changing this at all. 

So I switched the name to wyzesenseCLIENT and restarted the service - and all five devices that were paired previously was auto-discovered by Mosquito, and are currently working. "wyzesenseCLIENT" showed up one time in the Mosquito log however...

**What confuses me the most:**
Looking at the rest of the Mosquito log, I saw repeating series of entries like this, associated with the IP of the Pi (192.168.1.223) running wyzesense:

```
1601295122: New connection from 192.168.1.223 on port 1883.
[INFO] found kevinmqtt on local database
1601295124: New client connected from 192.168.1.223 as auto-66545BF5-22D0-CB71-DDF3-F85EF4C25A32 (p2, c1, k60, u'kevinmqtt').
1601295124: Client auto-66545BF5-22D0-CB71-DDF3-F85EF4C25A32 disconnected.

```

The "new" clientID I set in the config.yaml (wyzesenseCLIENT) _never shows up in this log_ except right after I first restarted and everything came online last night. Here is the Mosquito log, with all the auto client IDs. My rudimentary understanding of MQTT is that client ID is not required, and if its missing, one is auto generated every time a client connects: 

```
1601293324: New client connected from 192.168.1.223 as auto-F004F27F-3279-0CC6-4144-772F37C96DAB (p2, c1, k60, u'kevinmqtt').
1601293324: Client auto-F004F27F-3279-0CC6-4144-772F37C96DAB disconnected.
1601293382: New connection from 192.168.1.223 on port 1883.
1601293382: New client connected from 192.168.1.223 as auto-E9251CAF-9C02-9784-C6C3-5B5C14C318EB (p2, c1, k60, u'kevinmqtt').
1601293382: Client auto-E9251CAF-9C02-9784-C6C3-5B5C14C318EB disconnected.
1601293443: New connection from 192.168.1.223 on port 1883.
1601293443: New client connected from 192.168.1.223 as auto-273CDD72-0F31-753D-DA11-FEE543144106 (p2, c1, k60, u'kevinmqtt').
1601293443: Client auto-273CDD72-0F31-753D-DA11-FEE543144106 disconnected.
1601293503: New connection from 192.168.1.223 on port 1883.
1601293503: New client connected from 192.168.1.223 as auto-B75F1B78-22A2-0C30-4E4F-B5F998D9B7C1 (p2, c1, k60, u'kevinmqtt').
1601293503: Client auto-B75F1B78-22A2-0C30-4E4F-B5F998D9B7C1 disconnected.
1601293562: New connection from 192.168.1.223 on port 1883.
1601293562: New client connected from 192.168.1.223 as auto-382226B1-46D6-8D31-A11A-A891BF7E0F18 (p2, c1, k60, u'kevinmqtt').
1601293562: Client auto-382226B1-46D6-8D31-A11A-A891BF7E0F18 disconnected.
1601293623: New connection from 192.168.1.223 on port 1883.
1601293623: New client connected from 192.168.1.223 as auto-A91016DE-D3F8-A88E-94FE-81651A82C8B0 (p2, c1, k60, u'kevinmqtt').
1601293623: Client auto-A91016DE-D3F8-A88E-94FE-81651A82C8B0 disconnected.
1601293683: New connection from 192.168.1.223 on port 1883.
[INFO] found kevinmqtt on local database
1601293684: New client connected from 192.168.1.223 as auto-26B9C85A-7FB7-DA11-3704-ACD8DCFFF630 (p2, c1, k60, u'kevinmqtt').
1601293684: Client auto-26B9C85A-7FB7-DA11-3704-ACD8DCFFF630 disconnected.
1601293742: New connection from 192.168.1.223 on port 1883.
1601293742: New client connected from 192.168.1.223 as auto-D9D7E2FF-B8B3-5190-D36E-59DBF956CD06 (p2, c1, k60, u'kevinmqtt').
1601293742: Client auto-D9D7E2FF-B8B3-5190-D36E-59DBF956CD06 disconnected.
1601293803: New connection from 192.168.1.223 on port 1883.
1601293803: New client connected from 192.168.1.223 as auto-63BC8D34-2921-1643-F529-AC9B3D20E046 (p2, c1, k60, u'kevinmqtt').
1601293803: Socket error on client auto-63BC8D34-2921-1643-F529-AC9B3D20E046, disconnecting.
1601293863: New connection from 192.168.1.223 on port 1883.
1601293863: New client connected from 192.168.1.223 as auto-DE8FC0F7-3890-D23A-4E5D-0B7866BABD44 (p2, c1, k60, u'kevinmqtt').
1601293863: Client auto-DE8FC0F7-3890-D23A-4E5D-0B7866BABD44 disconnected.
1601293922: New connection from 192.168.1.223 on port 1883.
1601293922: New client connected from 192.168.1.223 as auto-614A321A-FB33-E3C1-C61E-45D8EE21E8F8 (p2, c1, k60, u'kevinmqtt').
1601293922: Client auto-614A321A-FB33-E3C1-C61E-45D8EE21E8F8 disconnected.
1601293983: New connection from 192.168.1.223 on port 1883.
1601293983: New client connected from 192.168.1.223 as auto-CF2195EB-5863-D52B-D7B5-76CD5BD3E351 (p2, c1, k60, u'kevinmqtt').
1601293983: Client auto-CF2195EB-5863-D52B-D7B5-76CD5BD3E351 disconnected.
1601294042: New connection from 192.168.1.223 on port 1883.
[INFO] found kevinmqtt on local database
1601294044: New client connected from 192.168.1.223 as auto-199593B6-44CC-D7CD-6A7E-50FF67BDAC2E (p2, c1, k60, u'kevinmqtt').
1601294044: Client auto-199593B6-44CC-D7CD-6A7E-50FF67BDAC2E disconnected.
1601294102: New connection from 192.168.1.223 on port 1883.
1601294102: New client connected from 192.168.1.223 as auto-656D4277-277E-15ED-8E4A-90119290990B (p2, c1, k60, u'kevinmqtt').
1601294102: Client auto-656D4277-277E-15ED-8E4A-90119290990B disconnected.
1601294163: New connection from 192.168.1.223 on port 1883.
1601294163: New client connected from 192.168.1.223 as auto-0D3034AF-EDD2-795A-5296-A4F187FF8E3D (p2, c1, k60, u'kevinmqtt').
1601294163: Client auto-0D3034AF-EDD2-795A-5296-A4F187FF8E3D disconnected.
[... cut ...]
1601295063: New client connected from 192.168.1.223 as auto-0E736EB5-6A46-9EE6-239C-D640DE235081 (p2, c1, k60, u'kevinmqtt').
1601295063: Client auto-0E736EB5-6A46-9EE6-239C-D640DE235081 disconnected.
1601295122: New connection from 192.168.1.223 on port 1883.
[INFO] found kevinmqtt on local database
1601295124: New client connected from 192.168.1.223 as auto-66545BF5-22D0-CB71-DDF3-F85EF4C25A32 (p2, c1, k60, u'kevinmqtt').
1601295124: Client auto-66545BF5-22D0-CB71-DDF3-F85EF4C25A32 disconnected.
```

PeterSelvaraj, 2020-10-06:
I took the docker route and I ran into the same issue couple of days ago. I repairing the devices did not fix the issue. So I decided to delete the docker images and start over. Downloading fresh images fixed the issue. I did not change anything in the configuration file. It continues to work without any issues.

AK5nowman, 2020-10-07:
@PeterSelvaraj did you try changing the mqtt client ID and restarting before reinstalling?

PeterSelvaraj, 2020-10-09:
@AK5nowman  No I didn't.

raetha, 2020-11-28:
A quick note I can add here is that I've occasionally had similar issues. Each time, I restart Mosquitto and things seem to solve themselves. I haven't taken the time to figure out if there is some sort of glitch in Mosquitto, Paho MQTT , or the way I'm using the Paho library to connect to MQTT. I believe I'm following "best practices" from Paho, but this stuff was brand new to me when I wrote it. So if anyone out there sees anything that looks off, let me know, or submit a PR. But you can probably correct this particular issue just by restarting Mosquitto.

dlashua, 2020-11-28:
Chiming in to say this happens to me quite often as well. I have two instances of wyze2mqtt running, both with different client_ids, both running in docker. The next time it happens, I'll try changing client_id to see if it helps.

dlashua, 2020-11-28:
As far as best practices for Paho MQTT goes, there are a few things to be aware of. 

1) Using LWT will allow an external observer to at least know that the service has disconnected.  This is pretty simple to do.

Before connecting use `MQTT_CLIENT.will_set(topic, payload='disconnected', qos=0, retain=False)`

After connecting publish 'connected' to that same topic. MQTT will handle the rest.

2) Paho will let you `publish` even if you're disconnected. It saves these and sends them later once a connection happens. In every case where you're publishing you can check `MQTT_CLIENT.is_connected()` to know if you're connected or not. As it is now, these look like successful publishes in the log, when they may not be. You could log whenever a publish happens and the client is disconnected. I would write a function called 'publish'. In it, you can perform the publish, log the published data and add a message in the log if MQTT is disconnected when publishing. Then instead of calling MQTT_CLIENT.publish, call your `publish` function instead. 

3) You're setting up the reconnect parameters, but never actually requesting a reconnect. In `on_disconnect` add `MQTT_CLIENT.reconnect()` to the bottom of that to start Paho's reconnect functionality. Once reconnected, all published messages will be sent, so you don't need to worry about them.

If these changes are agreeable to you, it's probably easier for you to write and test them since I don't have a spare WyzeSense dongle to test with. However, if needed, I can disable part of my home temporarily in order to code, test, and issue a PR for this.

AK5nowman, 2020-11-28:
I have a suspicion that it is Paho not respecting the client ID specified when it tries to reconnect. Haven't had the chance to troubleshoot this issue though as it hasn't happened after I added a bunch of logging

dlashua, 2020-11-29:
> I have a suspicion that it is Paho not respecting the client ID specified when it tries to reconnect. Haven't had the chance to troubleshoot this issue though as it hasn't happened after I added a bunch of logging

I just wrote a short test script that uses the Paho "reconnect" method. It respected the client_id.

You can see the [code and the logs in this gist](https://gist.github.com/dlashua/ffb17e3abcce125ea42da5edc6d4dcc2).

raetha, 2020-11-30:
@dlashua Appreciate the suggestions for improvement. Responses below by same number you used.

1) I'm not opposed to LWT, however I'd prefer the see the connection code more reliable, rather than relying on an external monitor looking for the disconnect topic and triggering restarts. Open to a PR on this though, just not high priority with my limited time.

2) I've just complete some tweaks to a few things here. First, removed an exit call in the on_connect code, think that may have been competing with loop_forever and killing the thread under certain circumstances. This might result in the MQTT connection being dead, but everything else appearing alive. Also took your suggestion on creating and internal publish function, which has some failure logging, which does capture if the connection is currently down. Lastly, I've set up code (currently commented) to switch to the loop_start and loop_stop methods for Paho. If we keep having issues with loop_forever, then things can quickly be switched, but it requires more long term code and handling, so I'd like to stick with loop_forever if we can, but do need to solve these issue if possible.

3) As I mentioned above, it currently uses the loop_forever function which handles all reconnection stuff automatically, and doesn't want any other loop or reconnection calls. Loop_start and loop_stop also appear to handle reconnections automatically. The only case I can find to call reconnect manually would be if we switched things to manually loop through the pending requests only when sending them. It appears there are a number of different use cases with Paho, but that not all functions are meant to work together. For example, they say don't use loop, loop_start/loop_stop, and loop_forever in the same script, just pick one model. Let me know if you've seen something different though around reconnect. It does seem fairly stable now, but there are definitely some conditions killing things, and I'm hoping some of these tweaks will help there.

dlashua, 2020-11-30:
@raetha excellent! I didn't realize the other methods automatically reconnected, so, TIL. 

I'll pull your new code now and see if it keep me from having to manually restart every few days. thanks!

tehstevo, 2020-12-02:
I had the same issues as OP.  Changing the mqtt_client_id kicked it in the pants and got the messages to update in HA.  The wyzesense2mqtt logs were showing the motion/contact the entire time.

dlashua, 2020-12-10:
@raetha I made it 10 days, which is a vast improvement from before. But, finally, today, both instances of wyze2mqtt that I run were no longer connected to MQTT, yet still running (and indicating in the logs with each attempted publish that it was not connected). 

Looking deeper in the past, I see other disconnections, retries, and eventual reconnections. But, this last time, it didn't manage to reconnect on it's own. 

A simple restart (I run wyzesense2mqtt in docker, so I just restarted the container) and everything is back up and running.

Perhaps a config option to not reconnect, but, instead, just exit? At least, this way, docker/supervisord could restart and, perhaps, it'll manage to recover on its own that way? I wish I could figure out WHY it eventually fails. There's nothing in the code that looks incorrect or out of place. So, I'm thinking it must be something in Paho.

AK5nowman, 2020-12-10:
@dlashua do your broker logs show anything funky going on during this period of time? Snippets of logs around the key points in time can be helpful as well. 

Curious, what is between your broker and wyzesense2mqtt? Are they on the same hardware device or separate? Mine are in separate VMs running on the same host and I have been running for months without this sort of issue now

dlashua, 2020-12-10:
I have two instances of wyzesense2mqtt running. One runs in docker on a host called "iceman". In that same docker I run mosquitto, zigbee2mqtt, and homeassistant, which all use MQTT. The is connected by wired LAN. The second instance of wyzesense2mqtt also runs in docker on a RaspberryPi 4 (called "goose") along with another instance of zigbee2mqtt and rtl_433tomqtt (all of which use the same MQTT server on the previously mentioned docker host). I have yet another docker host connected by wired LAN which runs a service called "viserson" that also connected to MQTT. I have 8 or so other wireless ESP devices that connect to MQTT. And my workstation which is connected by wired LAN run MQTT-Explorer fairly regularly.

Looking in the Mosquitto logs at one of these time periods, both wyzesense2mqtt clients disconnected, as did MQTT-Explorer. Everything else stayed connected.

The logs provide this timeline:

00:32:44 I see a PUBLISH from wyzesense2mqtt on iceman . This is preceded by MANY publishes at regular intervals so I have to assume it was working fine before this point. However, 00:32:44 is the LAST publish I see for a while.

00:45:56 I see a PUBLISH from wyzesense2mqtt on goose. Same as above, it's the LAST publish I see for a while.

01:21:59 wyze on goose logs that it connected to MQTT. I believe I restarted it at this time because I also see "Connecting to dongle." in the logs.

02:04:38 wyze on iceman connects to MQTT. Same as above, I believe this is when I restarted it.

02:05:29 mosquitto logs show wyze on both goose and iceman connecting. The timestamps I'm using come from docker. I think, perhaps, Mosquitto buffers its logs so, at this time point, it's outputting the logs from both of the previous connections. 

Some observations.

Since neither wyze instance shows "MQTT publish error: The client is not currently connected" until AFTER it reconnects to the dongle and all in the same second that it successfully connects to MQTT (likely after a manual restart), it's possible the host is losing connection with the wyze dongle. However, because it's happening on two different hosts, one wireless and one wired, one with an Intel processor and one a Raspberry Pi, it seems unlikely that it's host related. So, it's either IP network related or both of these dongles are receiving some RF that causes them to stop doing what they are supposed to do.

The wyzesense2mqtt code looks like it would continue to publish even when not connected to MQTT (and log that each time), so if the MQTT server just went away (due to network issues or whatever) I should still see some publish attempts. The largest gap I have in logs of "publish" during normal operation is about 45 minutes. It seems wyzesense2mqtt doesn't ever log a "disconnected from MQTT".

So, perhaps my issue is not MQTT related at all and, instead, has to do with the dongles.

That's progress! So thanks for having me dig through the logs. I'll keep sifting through them to see if I can find anything else that's interesting.

dlashua, 2020-12-10:
For further diagnosis here are my complete logs from  one host just before the outage until just after (though, possibly this needs a new thread as I'm not sure it's related to MQTT any more):

```
2020-12-10T00:32:44.727085671Z State event data: [2020-12-09 18:32:12][7793F663]StatusEvent: sensor_type=motion, state=inactive, battery=95, signal=84
2020-12-10T00:32:44.727526038Z {'event': 'status', 'available': True, 'mac': '7793F663', 'device_class': 'motion', 'last_seen': 1607560332.483, 'last_seen_iso': '2020-12-09T18:32:12.483000', 'signal_strength': -84, 'battery': 95, 'name': 'dance_motion', 'state': 0}
2020-12-10T00:32:44.733802966Z Received PUBCOMP (Mid: 675)
2020-12-10T00:32:44.733823769Z Sending PUBLISH (d0, q2, r1, m676), 'b'wyzesense2mqtt-iceman/7793F663'', ... (234 bytes)
2020-12-10T00:32:44.733827285Z Received PUBREC (Mid: 676)
2020-12-10T00:32:44.733829966Z Sending PUBREL (Mid: 676)
2020-12-10T00:32:44.733832674Z Received PUBCOMP (Mid: 676)
2020-12-10T00:33:44.809323439Z Sending PINGREQ
2020-12-10T00:33:44.810501873Z Received PINGRESP
2020-12-10T00:34:44.887805604Z Sending PINGREQ
2020-12-10T00:34:44.889018183Z Received PINGRESP
2020-12-10T00:35:44.962854515Z Sending PINGREQ
2020-12-10T00:35:44.964077224Z Received PINGRESP
2020-12-10T00:36:45.029961536Z Sending PINGREQ
2020-12-10T00:36:45.030591632Z Received PINGRESP

[ same thing every minute until .... ]

2020-12-10T02:00:51.371731249Z Sending PINGREQ
2020-12-10T02:00:51.373013831Z Received PINGRESP
2020-12-10T02:01:51.451888017Z Sending PINGREQ
2020-12-10T02:01:51.453052610Z Received PINGRESP
2020-12-10T02:02:51.530605543Z Sending PINGREQ
2020-12-10T02:02:51.531732968Z Received PINGRESP
2020-12-10T02:03:51.608012240Z Sending PINGREQ
2020-12-10T02:03:51.609354572Z Received PINGRESP

[ ... I restart wyzesense2mqtt ... ]

2020-12-10T02:04:30.617005185Z Logging initialized...
2020-12-10T02:04:30.617028632Z Initializing configuration...
2020-12-10T02:04:30.620767889Z Connecting to MQTT host mqtt.lan
2020-12-10T02:04:30.627497801Z Connecting to dongle /dev/hidraw0
2020-12-10T02:04:31.093969781Z State event data: [2020-12-09 20:00:15][77A64474]AlarmEvent: sensor_type=motion, state=inactive, battery=95, signal=80
2020-12-10T02:04:31.094057510Z Exception in thread Thread-1:
2020-12-10T02:04:31.094087314Z Traceback (most recent call last):
2020-12-10T02:04:31.094104006Z   File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
2020-12-10T02:04:31.096219770Z     self.run()
2020-12-10T02:04:31.096283689Z   File "/usr/local/lib/python3.8/threading.py", line 870, in run
2020-12-10T02:04:31.097299912Z     self._target(*self._args, **self._kwargs)
2020-12-10T02:04:31.097348989Z   File "/wyzesense2mqtt/wyzesense.py", line 383, in _Worker
2020-12-10T02:04:31.098154974Z     self._HandlePacket(pkt)
2020-12-10T02:04:31.098197600Z   File "/wyzesense2mqtt/wyzesense.py", line 358, in _HandlePacket
2020-12-10T02:04:31.098708903Z     handler(pkt)
2020-12-10T02:04:31.098752383Z   File "/wyzesense2mqtt/wyzesense.py", line 287, in _OnSensorAlarm
2020-12-10T02:04:31.099149959Z     self.__on_event(self, e)
2020-12-10T02:04:31.099180769Z   File "./wyzesense2mqtt.py", line 443, in on_event
2020-12-10T02:04:31.099826488Z     if (event.MAC not in SENSORS):
2020-12-10T02:04:31.099887009Z NameError: name 'SENSORS' is not defined
2020-12-10T02:04:33.094403097Z No device found on path /dev/hidraw0: _DoCommand
2020-12-10T02:04:33.094659054Z Reading sensors configuration...
2020-12-10T02:04:33.116012785Z Traceback (most recent call last):
2020-12-10T02:04:33.116033893Z   File "./wyzesense2mqtt.py", line 501, in <module>
2020-12-10T02:04:33.116100510Z     init_sensors()
2020-12-10T02:04:33.116109376Z   File "./wyzesense2mqtt.py", line 180, in init_sensors
2020-12-10T02:04:33.116178788Z     result = WYZESENSE_DONGLE.List()
2020-12-10T02:04:33.116187253Z NameError: name 'WYZESENSE_DONGLE' is not defined

[ ... this happens sometimes when I start wyzesense2mqtt. I just restart again and the issue resolves. So, I restart again ... ]
[ ... the above exception happens again. I restart again ... ]

2020-12-10T02:04:37.327406455Z Logging initialized...
2020-12-10T02:04:37.327433726Z Initializing configuration...
2020-12-10T02:04:37.330682790Z Connecting to MQTT host mqtt.lan
2020-12-10T02:04:37.333091529Z Connecting to dongle /dev/hidraw0
2020-12-10T02:04:37.898464817Z Dongle /dev/hidraw0: [ MAC: 7789771B, VER: 0.0.0.30 V1.4 Dongle UD3U, ENR: b'&\x94\xca\xe2\xb3\xfe<"\xa4\x9eZ\xe7\xe4\xdd\x88E']
2020-12-10T02:04:37.898840245Z Reading sensors configuration...
2020-12-10T02:04:38.112688965Z Linked sensors: ['7789B53F', '7788FCF1', '7788FED7', '7793F1D1', '77932C81', '7780005F', '7793FE52', '7788FCA5', '7788FD35', '77A64433', '77A64E54', '77A6596E', '77A65ACE', '7793F663', '77841D85', '77A64474', '77931CD8', '77A67C13', '77A66751', '77A677B1', '77A67323', '77845766', '77830C5F']
2020-12-10T02:04:38.113258767Z Publishing discovery topics for 7780005F
2020-12-10T02:04:38.114262235Z MQTT publish error: The client is not currently connected.
2020-12-10T02:04:38.114379766Z   homeassistant/binary_sensor/wyzesense_7780005F/state/config
2020-12-10T02:04:38.114722310Z   {"name": "bathkids_motion", "dev_cla": "motion", "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2mqtt-iceman/7780005F", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_7780005F_state", "stat_t": "wyzesense2mqtt-iceman/7780005F", "dev": {"identifiers": ["wyzesense_7780005F", "7780005F"], "m [... cut ...]
2020-12-10T02:04:38.207835705Z Sending PUBLISH (d0, q2, r1, m3), 'b'homeassistant/sensor/wyzesense_7780005F/battery/config'', ... (363 bytes)
2020-12-10T02:04:38.208400736Z Sending PUBLISH (d0, q2, r1, m4), 'b'homeassistant/binary_sensor/wyzesense_77830C5F/state/config'', ... (402 bytes)
2020-12-10T02:04:38.208771092Z Sending PUBLISH (d0, q2, r1, m5), 'b'homeassistant/sensor/wyzesense_77830C5F/signal_strength/config'', ... (394 bytes)
2020-12-10T02:04:38.209379699Z Sending PUBLISH (d0, q2, r1, m6), 'b'homeassistant/sensor/wyzesense_77830C5F/battery/config'', ... (360 bytes)
```

raetha, 2025-03-03:
@nabahr's recent PR included significant, and well tested, updates to the MQTT functions among other changes. Looking back over this issue it seems that most remaining things were MQTT stability related, hopefully now solved with the recent merge.

closed 2025-03-03 by raetha