raetha/wyzesense2mqtt #47: Add sensor_name to log entries

dennyfmn, 2020-11-19:
Now that we have the option to publish the sensor_name in the MQTT output, it would be really nice to be able to see the sensor_name in the log entries. When you have many sensors, it's hard to remember which one is which when viewing the log.

raetha, 2020-11-28:
I'll look into this, but it might require some tweaks that will slow things down, since I think I may need to pass more of the large dictionaries around to make the names more available. All the Wyze stuff operates purely on MAC, so that was being passed by itself. I like this idea though.

Are there any specific log messages where you would find this most handy? I might start there and see if there are any performance hits or other issues.

AK5nowman, 2020-11-28:
The only place I saw that writes the MAC in a log message was in wyzesense.py. I was thinking some sort of  "get_name" callback or something along those lines

raetha, 2020-11-30:
Hurm, yeah, I think we are mostly only writing the MACs to the log as debug messages coming from the WyzeSensePy library. Though we've got a modified version now, those changes once solidified will be PR'd back to the source library. So we really can't pull data back into it without breaking the ability to keep updating the root.

If there are specific messages you are seeing though, maybe we can log the sensor name on a separate line from the WyzeSensePy log entries. There are a few log entries that dump from the upstream calls to validate they are working. So those debug logs were more for development than users. We don't actually intend debug to be used on a regular basis.

Anyway, please paste some specific log lines that you are looking at, and we can be more specific and/or tweak those lines in particular.

dennyfmn, 2020-12-12:
Here are a couple of lines from the current log:

2020-12-11 19:25:32 INFO     wyzesense2mqtt  State event data: [2020-12-11 19:25:30][7785D1EA]StateEvent: sensor_type=switch, state=open, battery=86, signal=76
2020-12-11 19:25:38 INFO     wyzesense2mqtt  State event data: [2020-12-11 19:25:36][7785D1EA]StateEvent: sensor_type=switch, state=close, battery=86, signal=74

It would be very nice to see the name here.