raetha/wyzesense2mqtt #49: Contact Sensors Dying and Generation 2

dennyfmn, 2020-11-19:
I haven't been very careful about watching the battery levels in my (8) contact sensors. Wyze has a statement on their site that if you let the battery run completely down, the sensor may not function any more.

I am running the systemd version of the installation.

The first sensor would do nothing when I replaced the battery. No red LED blink when I installed the new battery. I called their support number and after going through their troubleshooting routine, they were kind enough to send me a replacement under warranty. I received it with a dead battery. It is functioning properly with a new battery.

The second sensor flashes the LED when  I install a new battery, but did not publish any MQTT messages. I tried to re-pair it with the gateway and found that it will not pair. The pairing function from the command line tool sees the sensor, but the MAC address is blank. The log says 'Invalid MAC address". I've gone around this loop quite a few times without success. The sensor flashes the LED when the magnet is moved, but will not pair with the gateway.

This makes me wonder if there is some memory in the device that looses its content after some time with a dead battery.

I wanted to buy some more contact sensors to replace the failed ones and found that they are not selling the current version anymore. They say watch for Generation 2 coming soon.

When I was talking to their support guy on the phone, I asked him about this. He had no concrete answers about when Generation 2 would be available, or whether it would be backwards compatible with the current sensors. He did say they will have a humidity sensor. He also said the the new sensors might need a new hub.

Does anyone have any more information about this? I'm hoping that Generation 2 will be compatible with what we have now.

AK5nowman, 2020-11-25:
@dennyfmn the loss of the MAC address seems to be a fairly common issue with no fix. If the logs indicate it is an invalid mac address it is because it lost its MAC and is all 0s. I've added support for some "status" messages, which seem to periodically report its state when it hasn't been activated in a fixed amount of time. This will let you alarm on sensor battery level even if it doesn't alarm regularly.  See #50

raetha, 2020-11-28:
Hi @dennyfmn and welcome! I don't have concrete answers for you, not actually associated with Wyze, but I'll share what I think might happen.

RE Sense v2. I believe there is a chance that the new sensors will be backwards compatible, based on Neos using a similar protocol, but already having a leak/humidity sensor. That said, Wyze could be going in a completely new direction, I have seen things from them about a new hub, and the v3 Cam not supporting the current gen stuff. So this may be a start from scratch situation. I'm waiting to see before declaring any intent with this project. But I don't have as much time as I did when I started, so if it is all new, someone else might need to take up the mantle to get this working with both generation products.

RE the dead sensors. I've definitely had a bunch that have gone to the invalid MAC state. However, in all cases (except one that I physically broke while changing the battery), if I took out the low battery and let them sit for a bit, then install a fresh one, they all got their MAC back and resynced to the bridge. I might have gotten lucky, but I don't think letting the battery get low is a death sentence, just requires patience to get going again. Because the bad MACs wreck havoc on the bridge stability, I've tried to block them as much as possible in code, thus the log messages. If you somehow do still have one linked, you should be able to clear it via the Fix function in the command line tool.

RE the battery alarms, @AK5nowman did a great job getting the base in place. As soon as I can, I'm going to look into specifically HA's handling of MQTT messages and see if I can turn the alarms into something actionable.

In the interim, I'm going to close this specific issue, but if you have other questions or issues come up, please feel free to create more.

closed 2020-11-28 by raetha