raetha/wyzesense2mqtt #52: Generation 2 Sensors

drinfernoo, 2021-01-28:
As many others have reported, I have a handful of sensors that have apparently dropped their MAC, and become unresponsive.

I've proactively spoken with a representative at Wyze, who has agreed to replace my sensors at no cost (and with an unknown shipping timeframe, at this time 🙄), but I'm also intending on purchasing their home monitoring "Core Starter Kit", which includes a few of the new sensors, and the hub. It also includes a Sense keypad, which would be a new device in the mix.

According to the representative I spoke with, the new sensors and keypad will only be compatible with the new hub, so I expect that any existing third-party integrations (like this one) will need to be reworked for compatibility...

When I get mine in (maybe not until March, as they say "Ships February 2021), I'd like to offer to help further this project, if at all possible... but if you have the motivation and time needed to invest more into this project, I may be inclined to furnishing you with the same setup.

Would you (potentially) be open to this arrangement?

AK5nowman, 2021-02-19:
There are a lot of unknowns right now regarding version two of the sensors. When they say version 2 sensors aren’t compatible with the version 1 dongle do they mean software wise, and that they are choosing not to support it. Or do they mean hardware wise, in that it’s a different protocol all together. The most telling would be if version 1 sensors were compatible with the version 2 hub, but I haven’t seen an answer on that yet. 

If it’s not compatible with the version 1 in any way chances are this project won’t be of any help. And it would be more difficult to intercept data packets because it’s a standalone hub device.

drinfernoo, 2021-02-19:
@AK5nowman True. The rep I spoke with ensured me that the *old* sensors would work with the new hub, but there's really no way of knowing.

I spoke with a different rep this morning who confirmed that they are having manufacturing issues, causing preorders to be delayed until April... sometime. A little bit irritated at that, as they were originally advertised as shipping in February 🤦🏽‍♂️

AK5nowman, 2021-02-19:
Well that brings some hope! 

Wow yeah your right, it was supposed to be February. I’ve brainwashed myself to thinks April is the original date so I don’t get sad... ha

drinfernoo, 2021-02-20:
From this article: https://www.androidpolice.com/2020/12/08/wyze-announces-60-year-home-monitoring-service-coming-in-march/ (admittedly from December 2020)

The Hub apparently has connectivity with: Ethernet, WiFi 802.11 b/g/n, 2.4GHz, Bluetooth 4.1+ BLE 5.1, **915MHz RF**

That last one is a Zigbee frequency...

AK5nowman, 2021-02-20:
Same as Version 1. If V1 will connect to their v2 hub it almost certainly won't be zigbee.

raetha, 2021-02-21:
I've been a bit radio silent on things for a bit due to work life in healthcare being crazy this past year. I don't expect I'll personally have time to modify things for the v2 sensors. But I'd love help from others to keep this project alive and supporting the new stuff if possible.

So if the two of you are willing, I will definitely accept pull requests, and honestly if you have the time and are willing, would be happy to add as a contributor on the main project so that releases don't need to wait on my time either.

Let me know your thoughts as we get closer to the new gear being available. I still wish Wyze would at least sell it without having to sign up for the monitoring service, as I'd be much more likely to at least play with it some and see what came of it.

AK5nowman, 2021-03-31:
Saw the FCC filings, looks like version 2 is based on the same chip so that is a positive thing to hear

AK5nowman, 2021-04-21:
Got my package in the mail today. So far I've had success pairing the v2 contact sensors (Sensor Type 14)  and the v2 Motion sensor (Sensor Type 15) to the v1 dongle. When I was pairing the keypad there was some activity and it verbally told me that pairing was successful but I'm not receiving any other data outside of that so far. Could very well be additional data packet types that the v1 dongle doesn't support. But good news for the new contact and motion sensors.

drinfernoo, 2021-04-21:
@AK5nowman That *is* good news! So are you not using the V2 hub for any part of it then? I haven't received mine yet, which is... frustrating 😅

AK5nowman, 2021-04-21:
@drinfernoo yeah, right now I have no intention of using the v2 hub unless there is a hack to use it without using the wyze servers.

AK5nowman, 2021-04-28:
I was able to take the firmware from the hms hub and flash it onto the V1 bridge. I now see the additional packets needed to support the keypad. Just need to finish reversing all the packet structures.

drinfernoo, 2021-05-07:
@AK5nowman, my stuff finally showed up! What can I do to help?

AK5nowman, 2021-05-07:
@drinfernoo thats great. You should be able to pair the motion and contact sensors right now without edit. I’m currently working on digging into the keypad. I think I’ve handled all the events from the keypad, just need to figure out the packets to get the keypad to speak and alarm. Also, currently doing this in c#, as I’m most comfortable with it. Once I get it figured out I’ll need some help making the changes to this python lib

drinfernoo, 2021-05-07:
@AK5nowman We also probably want to figure out communication with the new Hub, right? That would free up a USB port (from the bridge), and I'd assume it would also increase the signal strength with the sensors. We could also try and figure out how to use the siren!

AK5nowman, 2021-05-08:
@drinfernoo that ultimate would be to install a custom app to run on the T31 chip and have that act as a wyze sense to MQTT bridge. Even if we did have the SDK and developed that app I wouldn't know how to install it and get it to run. Using steps from https://github.com/HclX/WyzeHacks/issues/111#issuecomment-824558304 I have root access but besides pulling files I don't know what else is possible. Thought about perhaps trying to modify their dongle and alarm rule app to connect to a local endpoint and perhaps emulating their api calls.

Anyway, I pushed my repo to github [Here](https://github.com/AK5nowman/WyzeSense)

drinfernoo, 2021-05-08:
Well, I have to say that seems to go over my head 😅 I'm also not particularly keen in hardware modding my hub for the time being... But I could at least take a look at the C# project in the near future and see if I can help with getting it translated to Python, if that would help?

drinfernoo, 2021-05-09:
I've attempted to add one of my new motion sensors to my V1 bridge, and get the following state published:
```json
{
    "event": "alarm",
    "available": true,
    "mac": "77C006C8",
    "device_class": null,
    "last_seen": 1620522395.312,
    "last_seen_iso": "2021-05-08T18:06:35.312000",
    "signal_strength": -22,
    "battery": 102,
    "name": "Wyze Sense 77C006C8",
    "state": 0
}
```

I changed the info in `sensors.yaml` for this sensor:
```yaml
77C006C8:
  class: motion
  invert_state: false
  name: Front Door Motion
  sw_version: 23
```

`sw_version: 23` was filled in for me already. After changing this I published to `wyzesense2mqtt/reload` (as well as restarting the container, and even though the entities show up in HA, I am not getting any updates from it.

drinfernoo, 2021-05-09:
Actually, I do seem to be getting some state reporting from these, but not ever an actual state change...

```bash
State event data: [2021-05-08 21:46:02][77C006C8]AlarmEvent: sensor_type=unknown, state=unknown, battery=99, signal=24
{'event': 'alarm', 'available': True, 'mac': '77C006C8', 'device_class': None, 'last_seen': 1620535562.267, 'last_seen_iso': '2021-05-08T21:46:02.267000', 'signal_strength': -24, 'battery': 99, 'name': 'Front Door Motion', 'state': 0}
```

AK5nowman, 2021-05-09:
To be honest I never tested this library. I can take a look tomorrow and submit a PR

drinfernoo, 2021-05-14:
Well, the PR we're working on does seem to have fixed all of the functionality of the V2 **sensors**, but the keypad is still a mystery to me.

Are you flashing custom firmware to your V1 bridge, or your V2 hub?

AK5nowman, 2021-05-14:
I grabbed the firmware file from the HMS hub (using these steps to get root access https://github.com/HclX/WyzeHacks/issues/111#issuecomment-824558304) and flashed it to my V1 bridge using [WyzeSenseUpgrade](https://github.com/AK5nowman/WyzeSense/). This added the functionality needed to support the keypad data packets. Unfortunately you do lose control of the LED on the bridge so it is always yellow (and you can't toggle it on and off, though not supported in this lib).

raetha, 2021-05-14:
@AK5nowman, I likely wont be getting any of the v2 stuff until/unless Wyze starts selling it without requiring the hub purchase and "subscription". So I'm not able to test much here and validate any time soon. Since you seem to be pretty active on this stuff, would you be interested in being a collaborator on this the WS2MQTT project? That would let you update the master branch as you get stuff working, without having to wait on me finding time. I'm very happy to have this project live on, just not sure how much more I'll be able to contribute directly in the near future.

AK5nowman, 2021-05-16:
@raetha yeah that would be alright

drinfernoo, 2021-05-21:
So I've now successfully flashed my V1 bridge with the V2 Hub firmware from @AK5nowman, but just need to figure out how to receive things properly.

AK5nowman, 2021-05-21:
I believe my lib handled all the new packet types/subtypes to receive keypad modes and codes. Just hadn’t figured out how to trigger the alarm/other audio as of yet.

drinfernoo, 2021-05-21:
It does seem that yours reports everything out to me via the console, but it would still be ideal if we could get it working with WS2MQTT, so that we can integrate it easily into HA as an [Alarm Control Panel](https://www.home-assistant.io/integrations/alarm_control_panel.mqtt/).

I've done a quick logging hack, in order to get the payload from a pairing event in WS2MQTT:
` b'\x00\x00\x01y\x90\xcds\xc9\xa177C074BF\x05\x10\x9b\x00\x01\x00\x00\x00\x1f'`
But I'm not really sure what to do with it, or how to get the other events 🤔

AK5nowman, 2021-05-22:
Oh that is neat. The only issue is the “mode” and “pin” events are separate events. It seems like there is an event to “start a new instance” of a code/mode entry. We would have to hold the desired state in memory until a pin is received and then package it up and send it, unless of course it is the panic button.

drinfernoo, 2021-05-22:
@AK5nowman Yeah, as this is not just a sensor, but a keypad with multiple kinds of events, there will probably be quite a bit of different logic now.

RonSpawnson, 2021-07-29:
@drinfernoo I have v2 motion and contact sensors - did anything come of the PR for adding support for these v2 sensors? Need any additional testing?

drinfernoo, 2021-07-29:
@RonSpawnson The keypad is still in limbo (have not looked at it in awhile, not sure about @AK5nowman), but this project does now support the V2 contact and motion sensors. I have a few of each in use alongside my V1 ones, all paired to the V1 bridge and running wyzesense2mqtt.

RonSpawnson, 2021-07-29:
Interesting - my v2 door sensor does not appear to be pairing for some reason, my other sensors are pairing and working just fine. Puzzling. 

I'm using HA mqtt.publish to topic wyzesense2mqtt/scan and then pressing and holding on the pair button of my v2 contact sensor until the light blinks 3 times. But still I am not seeing an entity or events for this v2 contact sensor. It was working with ha-wyzesense project, so I don't believe the sensor is faulty.

AK5nowman, 2021-07-29:
@RonSpawnson - so all of these contact sensors that you tried using, were they paired and working prior to using wyzesense2mqtt?

RonSpawnson, 2021-07-29:
One of the v2 contact sensors was paired prior to wyzesense2mqtt (I had paired it a while back via kevinvincent ha-wyzesense custom HA integration), and it was the one I had to unpair and re-pair. I tried pairing via CLI too, and it still didn't work until I first unpaired with CLI tool and then paired again.

The other v2 contact sensor I tried was never paired previously. I think I was able to pair this just fine, though interestingly I had to pair it via CLI rather than the MQTT way working. I do not think I had to "unpair" this one. I have a third v2 sensor that was never-before paired that I could test with again to validate my theory here. This would seem to imply that how I was attempting to initiate pairing via MQTT wasn't working for some reason? I'll re-test again tonight with that 3rd v2 contact sensor to double check.

AK5nowman, 2021-07-29:
I’m curious if it’s actually entering scan mode or not. If you wouldn’t mind could you ensure you have logs enabled then trying pairing your last sensor. The logs might give us a hint

RonSpawnson, 2022-06-05:
Can someone confirm if the v2 hub is supported in this repo? I was thinking of picking one up to see if it fixes the issue where all my sensors stop working every few weeks and require restart of hub.

raetha, 2025-03-04:
@RonSpawnson sorry for massive delay, but as far as I know the V2 hub is networked directly, rather than being a USB dongle. I haven't seen anything that implies there is a way to directly talk to the hub, it works via the Wyze cloud. That said, if you've found something where we can talk to hub over the network, that would make it feasible to have this project utilize the newer hub. No idea what that would actually take, and my time is limited as is obvious from the radio silence for so long. Personally I'd love to see being able to replace the v1 dongles with the hub and still not require cloud.

AK5nowman, 2025-03-04:
I think I killed my HMS hub but I wonder if we could take some like [thingino](https://github.com/themactep/thingino-firmware) and compile it for the hub. If so it would likely be possible to interface with the onboard radio

raetha, 2025-03-04:
I would love to see alternate firmware for the hub. Though at that point it would probably make sense to run something like this directly on the hub so it communicated directly to mqtt rather than needing an intern intermediary server.

And honestly, I’d love to see an ESP32 based open source hardware that could talk to the sensors. Open hardware would allow this stuff to never die. But I don’t have those skills at all.

raetha, 2025-11-12:
I'm going to close this particular issue as at this point v2 sensor support is in place except for the keypad which is being worked on as PR #63. So I think we can call the spirit of things complete and just hope that those with keypads can finish the testing required to enable them soonish!