raetha/wyzesense2mqtt #57: 'user_config' referenced before assignment

RonSpawnson, 2021-06-19:
## Describe the Bug

Issues installing the script - when I start up the container the logs say the following: 

Copying default logging config file...
Logging initialized...
Initializing configuration...
Traceback (most recent call last):
  File "./wyzesense2mqtt.py", line 484, in <module>
    init_config()
  File "./wyzesense2mqtt.py", line 104, in init_config
    if (CONFIG != user_config):
UnboundLocalError: local variable 'user_config' referenced before assignment

### Steps To Reproduce

I created the config.yaml file and the directory that I put it in was referenced in my docker-compose.yml file when binding the volume mounts. But when I perform docker-compose up -d and check the logs I see the error referenced above.

### Expected Behavior

Container starts up.

Can anyone assist me in debugging what I'm doing wrong on install with docker method?

drinfernoo, 2021-07-29:
Probably need to see the relevant portion of your `docker-compose.yaml` in order to say if the volumes are correct.

RonSpawnson, 2021-07-29:
Thanks for the response - I'll be closing this issue as I upgraded to an Intel NUC and installed wyzesense2mqtt on a debian VM and it's working just fine on that.

closed 2021-07-29 by RonSpawnson