raetha/wyzesense2mqtt #58: Sensors stopped picking up events

RonSpawnson, 2021-08-01:
## Describe the Bug

After a few days of running the project, my sensors no longer seem to be picking up the events. I see MQTT pings in the logs, but no more sensor events after a particular time, even when I trigger motion or contact sensor. 

Logs from the last period in which motion events were ocurring:

```
Sending PUBLISH (d0, q2, r1, m5991), 'b'wyzesense2m2qtt/77A5F5DD'', ... (240 bytes)
Received PUBREC (Mid: 5991)
Sending PUBREL (Mid: 5991)
Received PUBCOMP (Mid: 5991)
State event data: [2021-07-31 21:12:15][77A5E051]AlarmEvent: sensor_type=motion, state=active, battery=96, signal=67
{'event': 'alarm', 'available': True, 'mac': '77A5E051', 'device_class': 'motion', 'last_seen': 1627783935.389, 'last_seen_iso': '2021-07-31T21:12:15.389000', 'signal_strength': -67, 'battery': 96, 'name': 'Wyze Sense 77A5E051', 'state': 1}
Sending PUBLISH (d0, q2, r1, m5992), 'b'wyzesense2m2qtt/77A5E051'', ... (240 bytes)
Received PUBREC (Mid: 5992)
Sending PUBREL (Mid: 5992)
Received PUBCOMP (Mid: 5992)
State event data: [2021-07-31 21:12:46][77C3CBD4]AlarmEvent: sensor_type=motionv2, state=inactive, battery=100, signal=70
{'event': 'alarm', 'available': True, 'mac': '77C3CBD4', 'device_class': 'motion', 'last_seen': 1627783966.929, 'last_seen_iso': '2021-07-31T21:12:46.929000', 'signal_strength': -70, 'battery': 100, 'name': 'Wyze Sense 77C3CBD4', 'state': 0}
Sending PUBLISH (d0, q2, r1, m5993), 'b'wyzesense2m2qtt/77C3CBD4'', ... (241 bytes)
Received PUBREC (Mid: 5993)
Sending PUBREL (Mid: 5993)
Received PUBCOMP (Mid: 5993)
State event data: [2021-07-31 21:12:47][77A8A248]AlarmEvent: sensor_type=motion, state=inactive, battery=96, signal=76
{'event': 'alarm', 'available': True, 'mac': '77A8A248', 'device_class': 'motion', 'last_seen': 1627783967.265, 'last_seen_iso': '2021-07-31T21:12:47.265000', 'signal_strength': -76, 'battery': 96, 'name': 'upstairs_hallway_motion_mqtt', 'state': 0}
Sending PUBLISH (d0, q2, r1, m5994), 'b'wyzesense2m2qtt/77A8A248'', ... (249 bytes)
Received PUBREC (Mid: 5994)
Sending PUBREL (Mid: 5994)
Received PUBCOMP (Mid: 5994)
State event data: [2021-07-31 21:12:47][77A8A248]AlarmEvent: sensor_type=motion, state=inactive, battery=96, signal=76
{'event': 'alarm', 'available': True, 'mac': '77A8A248', 'device_class': 'motion', 'last_seen': 1627783967.265, 'last_seen_iso': '2021-07-31T21:12:47.265000', 'signal_strength': -76, 'battery': 96, 'name': 'upstairs_hallway_motion_mqtt', 'state': 0}
Sending PUBLISH (d0, q2, r1, m5995), 'b'wyzesense2m2qtt/77A8A248'', ... (249 bytes)
Received PUBREC (Mid: 5995)
Sending PUBREL (Mid: 5995)
Received PUBCOMP (Mid: 5995)
State event data: [2021-07-31 21:12:47][77A5F4FA]AlarmEvent: sensor_type=motion, state=inactive, battery=94, signal=76
{'event': 'alarm', 'available': True, 'mac': '77A5F4FA', 'device_class': 'motion', 'last_seen': 1627783967.388, 'last_seen_iso': '2021-07-31T21:12:47.388000', 'signal_strength': -76, 'battery': 94, 'name': 'Wyze Sense 77A5F4FA', 'state': 0}
Sending PUBLISH (d0, q2, r1, m5996), 'b'wyzesense2m2qtt/77A5F4FA'', ... (240 bytes)
Received PUBREC (Mid: 5996)
Sending PUBREL (Mid: 5996)
Received PUBCOMP (Mid: 5996)
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
```

The log is then further filled with nothing other than PINGREQ and PINGRESP.

RonSpawnson, 2021-08-01:
More data - I'm trying to launch the CLI to debug, and it appears the device is not showing up on /dev/hidraw0 anymore. I also tried /dev/hidraw1 /dev/hidraw2 etc. with no luck. Something must be causing the device to be unrecognizable after a period of time?

```
python3 bridge_tool_cli.py --device /dev/hidraw0
Openning wyzesense gateway [/dev/hidraw0]
No device found on path /dev/hidraw0
```

RonSpawnson, 2021-08-02:
I've left the system in this state to try to debug (and will continue to do so for the next day or two), so if there things you would recommend for me to try or gather some data, I'd be happy to do so.

RonSpawnson, 2021-08-02:
```
sudo dmesg | grep hidraw

[    2.093750] hidraw: raw HID events driver (C) Jiri Kosina
[    2.337035] hid-generic 0003:0627:0001.0001: input, hidraw0: USB HID v0.01 Mouse [QEMU QEMU USB Tablet] on usb-0000:00:01.2-1/input0
[    2.339417] hid-generic 0003:1A86:E024.0002: hiddev0,hidraw1: USB HID v1.00 Device [HID 1a86:e024] on usb-0000:01:1b.0-1/input0
```

My understanding is that HID 1a86:e024 is the wyzesense dongle. It appears it is saying it is on hidraw1, which matches my expectations, as that is what I use in my docker-compose.yaml.

```
{...}
devices:
  - "/dev/hidraw1:/dev/hidraw1"
{...}
```

Yet when I try CLI for hidraw1 since this issue started happening (it randomly crashed) I see the following:

```
python3 bridge_tool_cli.py --device /dev/hidraw1
Openning wyzesense gateway [/dev/hidraw1]
No device found on path /dev/hidraw1
```

RonSpawnson, 2021-08-02:
Looking into system logs around the time this stopped working and I'm finding some more useful information.

```
Jul 28 22:57:43 dockerlab qemu-ga: info: guest-shutdown called, mode: (null)
Jul 28 22:57:43 dockerlab systemd[1]: unattended-upgrades.service: Succeeded.
Jul 28 22:57:43 dockerlab systemd[1]: systemd-rfkill.socket: Succeeded.
Jul 28 22:57:43 dockerlab systemd[1]: Closed Load/Save RF Kill Switch Status /dev/rfkill Watch.
Jul 28 22:57:43 dockerlab systemd[1]: Removed slice system-getty.slice.
Jul 28 22:57:43 dockerlab systemd[1]: Stopping Session 780 of user {REDACTED}.
Jul 28 22:57:43 dockerlab systemd[1]: Stopping Daemon for power management...

{lots more lines in between}

Jul 28 22:58:06 dockerlab kernel: [    2.093750] hidraw: raw HID events driver (C) Jiri Kosina
Jul 28 22:58:06 dockerlab kernel: [    2.100229] usbcore: registered new interface driver usbhid
Jul 28 22:58:06 dockerlab kernel: [    2.100230] usbhid: USB HID core driver
Jul 28 22:58:06 dockerlab kernel: [    2.336631] input: QEMU QEMU USB Tablet as /devices/pci0000:00/0000:00:01.2/usb1/1-1/1-1:1.0/0003:0627:0001.0001/input/input5
Jul 28 22:58:06 dockerlab kernel: [    2.337007] usb 2-1: New USB device found, idVendor=1a86, idProduct=e024, bcdDevice= 1.0e
Jul 28 22:58:06 dockerlab kernel: [    2.337009] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Jul 28 22:58:06 dockerlab kernel: [    2.337035] hid-generic 0003:0627:0001.0001: input,hidraw0: USB HID v0.01 Mouse [QEMU QEMU USB Tablet] on usb-0000:00:01.2-1/input0
Jul 28 22:58:06 dockerlab kernel: [    2.339417] hid-generic 0003:1A86:E024.0002: hiddev0,hidraw1: USB HID v1.00 Device [HID 1a86:e024] on usb-0000:01:1b.0-1/input0
```

It appears to me like something (unknown as to what) caused my VM to reboot, and further, that rebooting it caused my USB device to no longer be accessible (also unknown as to why this wouldn't survive a VM re-boot).

RonSpawnson, 2021-08-03:
I've rebooted the VM and still experienced this issue. I rebooted the NUC and am now back up and running, but I worry it will break again in a few days with same cause as I don't know what caused it. I'll monitor and report back if and when it crashes again.

RonSpawnson, 2021-08-14:
I think I might have found a fix! After upgrading my HA OS in another container, my wyzesense2mqtt stopped working. I tried restarting the container "docker restart wyzesense2mqtt" but it failed because it said there no longer was a "/dev/hidraw1". In my digging I happened across this interesting discussion about "dynamically connected USB devices" being available to docker containers. https://forums.balena.io/t/docker-container-cannot-access-dynamically-plugged-usb-devices/4277/13

I tried updating my docker-compose file with the following changes:

```
privileged: true
environment:
  - UDEV=1
  - TZ=America/Chicago
```

Then I tried "docker-compose up -d" and everything started working again.

RonSpawnson, 2021-08-30:
Hello - after a couple weeks, my sensors stopped picking up events again. All I see following this event is endless list of PINGREQ and PINGRESP.

```
State event data: [2021-08-28 17:32:50][77A5F5DD]AlarmEvent: sensor_type=motion, state=inactive, battery=94, signal=66
{'event': 'alarm', 'available': True, 'mac': '77A5F5DD', 'device_class': 'motion', 'last_seen': 1630189970.079, 'last_seen_iso': '2021-08-28T17:32:50.079000', 'signal_strength': -66, 'battery': 94, 'name': 'Wyze Sense 77A5F5DD', 'state': 0}
Sending PUBLISH (d0, q2, r1, m22393), 'b'wyzesense2m2qtt/77A5F5DD'', ... (240 bytes)
Received PUBREC (Mid: 22393)
Sending PUBREL (Mid: 22393)
Received PUBCOMP (Mid: 22393)
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
```

Any other logs or data I can provide to help track this down?

nabahr, 2021-09-14:
I've been having this issue for a while as well. I can usually get it running by restarting the service though. My intermediate fix is an automation that makes an ssh call to restart the service if it hasn't seen any MQTT events from the service in the last hour. With the amount of contact and motion sensors I have, we rarely go that long without triggering one of them.
It used to happen much more frequently, around once a week. Now it only happens about once a month or so.

AK5nowman, 2021-09-23:
@nabahr I’m curious if you are using docker as well?

nabahr, 2021-09-23:
I am running the service in a proxmox VM (Debian 10), along with zigbee2mqtt. I haven't had any issues with my zigbee service so far. Here is my VM config.
```boot: order=scsi0;net0
cores: 1
memory: 1024
name: MQTT
net0: virtio=1A:AA:D5:C9:8C:BC,bridge=vmbr0,firewall=1
numa: 0
onboot: 1
ostype: l26
scsi0: local-lvm:vm-105-disk-0,size=32G
scsihw: virtio-scsi-pci
smbios1: uuid=217a1af3-cceb-499a-a53a-681b00b18699
sockets: 1
usb0: host=1a86:e024,usb3=1
usb1: host=1cf1:0030,usb3=1
vmgenid: ccb5b5f4-3088-4634-ab84-5ad7c50aa8e3```

AK5nowman, 2021-09-24:
@nabahr and @RonSpawnson I've not been unfortunate enough to run into this issue lately. Next time it happens can you publish to the reload MQTT topic so that we can force it to attempt to the dongle. I'm curious if it will crash/error out when it attempts to write the packet bytes.

RonSpawnson, 2021-09-24:
Just tried publishing reload topic. Saw a bunch of activity in the output, but still no events.

```
Received PUBLISH (d0, q0, r0, m0), 'wyzesense2m2qtt/reload', ...  (0 bytes)
In on_message_reload:
Reading sensors configuration...
Linked sensors: ['MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED', 'MAC_OBFUSCATED']
Publishing discovery topics for MAC_OBFUSCATED
Sending PUBLISH (d0, q2, r1, m94), 'b'homeassistant/binary_sensor/wyzesense_MAC_OBFUSCATED/state/config'', ... (400 bytes)
  homeassistant/binary_sensor/wyzesense_MAC_OBFUSCATED/state/config
  {"name": "Wyze Sense MAC_OBFUSCATED", "dev_cla": null, "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_state", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Sending PUBLISH (d0, q2, r1, m95), 'b'homeassistant/sensor/wyzesense_MAC_OBFUSCATED/signal_strength/config'', ... (400 bytes)
  homeassistant/sensor/wyzesense_MAC_OBFUSCATED/signal_strength/config
  {"name": "Wyze Sense MAC_OBFUSCATED Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_strength }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_signal_strength", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Sending PUBLISH (d0, q2, r1, m96), 'b'homeassistant/sensor/wyzesense_MAC_OBFUSCATED/battery/config'', ... (366 bytes)
  homeassistant/sensor/wyzesense_MAC_OBFUSCATED/battery/config
  {"name": "Wyze Sense MAC_OBFUSCATED Battery", "dev_cla": "battery", "unit_of_meas": "%", "val_tpl": "{{ value_json.battery }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_battery", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Publishing discovery topics for MAC_OBFUSCATED
Sending PUBLISH (d0, q2, r1, m97), 'b'homeassistant/binary_sensor/wyzesense_MAC_OBFUSCATED/state/config'', ... (400 bytes)
  homeassistant/binary_sensor/wyzesense_MAC_OBFUSCATED/state/config
  {"name": "Wyze Sense MAC_OBFUSCATED", "dev_cla": null, "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_state", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Sending PUBLISH (d0, q2, r1, m98), 'b'homeassistant/sensor/wyzesense_MAC_OBFUSCATED/signal_strength/config'', ... (400 bytes)
  homeassistant/sensor/wyzesense_MAC_OBFUSCATED/signal_strength/config
  {"name": "Wyze Sense MAC_OBFUSCATED Signal Strength", "dev_cla": "signal_strength", "unit_of_meas": "dBm", "val_tpl": "{{ value_json.signal_strength }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_signal_strength", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Sending PUBLISH (d0, q2, r1, m99), 'b'homeassistant/sensor/wyzesense_MAC_OBFUSCATED/battery/config'', ... (366 bytes)
  homeassistant/sensor/wyzesense_MAC_OBFUSCATED/battery/config
  {"name": "Wyze Sense MAC_OBFUSCATED Battery", "dev_cla": "battery", "unit_of_meas": "%", "val_tpl": "{{ value_json.battery }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_battery", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Publishing discovery topics for MAC_OBFUSCATED
Sending PUBLISH (d0, q2, r1, m100), 'b'homeassistant/binary_sensor/wyzesense_MAC_OBFUSCATED/state/config'', ... (400 bytes)
  homeassistant/binary_sensor/wyzesense_MAC_OBFUSCATED/state/config
  {"name": "Wyze Sense MAC_OBFUSCATED", "dev_cla": null, "pl_on": "1", "pl_off": "0", "json_attr_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "val_tpl": "{{ value_json.state }}", "uniq_id": "wyzesense_MAC_OBFUSCATED_state", "stat_t": "wyzesense2m2qtt/MAC_OBFUSCATED", "dev": {"identifiers": ["wyzesense_MAC_OBFUSCATED", "MAC_OBFUSCATED"], "manufacturer": "Wyze", "model": "Sense Contact Sensor", "name": "Wyze Sense MAC_OBFUSCATED", "sw_version": ""}}
Sending PUBLISH (d0, q2, r1, m101), 'b'homeassistant/sensor/wyzesense_MAC_OBFUSCATED/signal_strength/config'', ... (400 bytes)
  homeassistant/sensor/wyzesense_MAC_OBFUSCATED/signal_strength/config
[... cut ...]
Received PUBCOMP (Mid: 154)
Received PUBCOMP (Mid: 155)
Received PUBCOMP (Mid: 156)
Received PUBCOMP (Mid: 157)
Received PUBCOMP (Mid: 158)
Received PUBCOMP (Mid: 159)
Received PUBCOMP (Mid: 160)
Received PUBCOMP (Mid: 161)
Received PUBCOMP (Mid: 162)
Received PUBCOMP (Mid: 163)
Received PUBCOMP (Mid: 164)
Received PUBCOMP (Mid: 165)
Received PUBCOMP (Mid: 166)
Received PUBCOMP (Mid: 167)
Received PUBCOMP (Mid: 168)
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP
Sending PINGREQ
Received PINGRESP

```

RonSpawnson, 2021-09-24:
New working theory - I'm noticing "Power - 1 new notification" and "Automatic suspend - computer will suspend very soon because of inactivity". I'm wondering if this causes it to lose USB access for some reason? I have now tried the fix detailed in https://askubuntu.com/a/1262605 and will report back if the issue reproduces again. If you don't hear back from me in the next 4 weeks or so it can be assumed this fixes the issue for proxmox users. Else, we are still looking for the cause/fix.

RonSpawnson, 2021-09-25:
Stopped picking up events again after a restart of HA due to software upgrade. One more theory - I still have the ha-wyzesense custom component. I wonder if when I restarted HA, ha-wyzesense custom component "took over" the wyze sense USB dongle. I'm going to remove the ha-wyzesense custom component and see if that resolves my issues.

RonSpawnson, 2021-09-28:
My sensors just stopped responding again even after the above attempted fixes :(

nabahr, 2021-11-05:
I have finally had the sensors stop reporting again and was able to try publishing a reload rather than restarting the service altogether and it did work! The sensors started reporting state again. I attached the relevant portion of the log, it was at around 16:35 when I noticed the sensors weren't triggering and issued the reload shortly after. I do see some errors later down in the log that might be of interest
[w2m_reload.log](https://github.com/raetha/wyzesense2mqtt/files/7489096/w2m_reload.log)
.

RonSpawnson, 2021-11-08:
I'm happy to report that after the combination of things I've tried above, I am now fully stable! I no longer have the sensors stopping publishing events issue. I'm unsure of which combination of factors solved this for me, but I do think the fact that I had the old HA wyzesense integration combating with this for the same USB sensor at the same time was likely a cause.

I'm going to close this issue as I seem to have been able to successfully fix my issue with the above steps, and now have a rock solid integration. Those still experiencing issues can experiment with some of the fixes I mentioned above, and let us know if any of them also solved your issue! If you are still having trouble, I encourage you to open your own ticket. Thanks for the help everyone!

RonSpawnson, 2022-01-17:
Unfortunately this issue is again occurring for me, it just only happens once every month or two. 

I tried transitioning my environment to running in a docker container on UNRAID to isolate any VM oddities. It ran well for a month but then the sensor events randomly stopped populating.  Restarting the container and it fails to start with the following in logs:

```
Logging initialized...
Initializing configuration...
Connecting to MQTT host 192.168.1.122
Connecting to dongle /dev/hidraw0
State event data: [2022-01-17 07:44:42][MAC OBSCURED]AlarmEvent: sensor_type=motion, state=active, battery=93, signal=76
Exception in thread Thread-1:
Traceback (most recent call last):
File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
self.run()
File "/usr/local/lib/python3.8/threading.py", line 870, in run
self._target(*self._args, **self._kwargs)
File "/wyzesense2mqtt/wyzesense.py", line 391, in _Worker
self._HandlePacket(pkt)
File "/wyzesense2mqtt/wyzesense.py", line 366, in _HandlePacket
handler(pkt)
File "/wyzesense2mqtt/wyzesense.py", line 295, in _OnSensorAlarm
self.__on_event(self, e)
File "./wyzesense2mqtt.py", line 440, in on_event
if (event.MAC not in SENSORS):
NameError: name 'SENSORS' is not defined
No device found on path /dev/hidraw0: _DoCommand
Reading sensors configuration...
Traceback (most recent call last):
File "./wyzesense2mqtt.py", line 498, in <module>
init_sensors()
File "./wyzesense2mqtt.py", line 187, in init_sensors
result = WYZESENSE_DONGLE.List()
NameError: name 'WYZESENSE_DONGLE' is not defined
```

I still see /dev/hidraw0 on the UNRAID CLI. No amount of restarts of the container fix this. I'm going to try restarting UNRAID to see if that resolves it.

RonSpawnson, 2022-01-17:
Update: A soft "restart" of UNRAID still didn't bring it back up. I had to physically power down the machine running UNRAID and bring it back up and then it started working again. Very odd. I'm curious what is happening when this happens and how we can avoid it?

timmerk, 2023-05-10:
This is happening for me, too. However, I'm running stock Homebridge on Debian 11 *without* Docker on a RaspPi 3+.

nabahr, 2023-05-10:
Since I opened this issue, I have since switched to a spare dongle that I had sitting in the drawer. The new dongle works way better than the old one. The old one has packet errors all the time, the new one very rarely has any packet errors (e.x. mismatched checksums).

The new dongle does, for whatever reason, still stop working maybe once every one or two months. The only way to get it back working stable is to unplug the dongle and plug it back in. I can get it working sometimes by just restarting the service, but it's not stable and stops working again after only a couple hours. At this point I am chalking it up to subpar firmware from wyze on the first gen devices, not a stretch considering the null mac issue with the sensors themselves.

timmerk, 2023-05-10:
@nabahr Thanks for the update! Unfortunately, I don't have that option, as my bridge is actually a Neos Smart Bridge. I just tried updating the firmware on it to the Wyze firmware that @AK5nowman posted on his repo (since they are very similar), but I bricked my bridge, hah. Oh well! It actually worked at first, then somehow bricked itself after a few minutes.

HclX, 2024-12-21:
hey, i started playing with this project recently, and ran into similar issue, which turned out to be an exception in the reader thread:

```
Trying to parse: b'55aa531d1900000193e7c6c812a1373743363831443007115a00010059ae4608c8'
Received: b'55aa531d1900000193e7c6c812a1373743363831443007115a00010059ae4608c8'
<=== Received: Packet: Cmd=5319, Payload=b'00000193e7c6c812a1373743363831443007115a00010059ae46'
Sending ACK packet for cmd 5319
===> Sending: Packet: Cmd=53FF, Payload=ACK(5319)
Sending: b'aa555319ff026a'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/wyzesense2mqtt/wyzesense.py", line 413, in _Worker
    self._HandlePacket(pkt)
  File "/wyzesense2mqtt/wyzesense.py", line 375, in _HandlePacket
    handler(pkt)
  File "/wyzesense2mqtt/wyzesense.py", line 285, in _OnSensorAlarm
    sensor_type = "unknown (" + type + ")"
TypeError: can only concatenate str (not "int") to str
Sending PUBLISH (d0, q2, r1, m220), 'b'wyzesense2mqtt/status'', ... (6 bytes)
Received PUBREC (Mid: 220)
Sending PUBREL (Mid: 220)
```
In my case, it's caused by the packet from climate sensor i paired. You can see this if you go check console output (the error messages are not in any log files).

raetha, 2025-03-03:
@HclX I'm back a bit and just pulled a lot of changes from @nabahr, as well as getting docker containers going again via GHCR. If you managed to fix the issue you mentioned in the last comment, I'd love to take a PR on that if it wasn't included with nahahr's PR. Trying to get things cleaned up, and hopefully all the new features implemented that people have put time into.

HclX, 2025-03-04:
Sure thing, I will need to take a look at what's included in the latest commit.

closed 2025-11-12 by raetha