raetha/wyzesense2mqtt #59: Publishing to `self_topic_root/reload` no longer works

drinfernoo, 2021-09-21:
## Describe the Bug

I added a new sensor today, and after modifying `sensors.yaml` to change the name and device class, tried publishing to `wyzesense2mqtt/reload`... But that just makes the container stop working. During this process, I was tailing the container log (`docker logs -f wyzesense2mqtt`), and nothing happened after the reload message was received, including any further state updates from the sensors. Restarting the container restored regular behavior.

### Steps To Reproduce

Steps to reproduce the behavior:

1. Publish a message to `self_topic_root/reload`

### Expected Behavior

Publishing a reload message should... reload the sensors 😅 

## Additional Context

The last bit of my log, and then nothing happened after this:
```bash
Sending PUBLISH (d0, q2, r1, m12685), 'b'wyzesense2mqtt/77C006C8'', ... (239 bytes)
Received PUBREC (Mid: 12685)
Sending PUBREL (Mid: 12685)
Received PUBCOMP (Mid: 12685)
State event data: [2021-09-20 17:15:40][77C00CE1]AlarmEvent: sensor_type=motionv2, state=active, battery=98, signal=72
{'event': 'alarm', 'available': True, 'mac': '77C00CE1', 'device_class': 'motion', 'last_seen': 1632183340.093, 'last_seen_iso': '2021-09-20T17:15:40.093000', 'signal_strength': -72, 'battery': 98, 'name': 'Back Porch Motion', 'state': 1}
Sending PUBLISH (d0, q2, r1, m12686), 'b'wyzesense2mqtt/77C00CE1'', ... (238 bytes)
Received PUBREC (Mid: 12686)
Sending PUBREL (Mid: 12686)
Received PUBCOMP (Mid: 12686)
Received PUBLISH (d0, q0, r0, m0), 'wyzesense2mqtt/reload', ...  (0 bytes)
In on_message_reload:
Reading sensors configuration...
```

AK5nowman, 2021-09-24:
@drinfernoo Can you check to ensure the formatting on your sensors.yaml file is correct? Though I would expect this to print an error...hmm.
Try adding the following before [181](https://github.com/raetha/wyzesense2mqtt/blob/master/wyzesense2mqtt/wyzesense2mqtt.py#L181)
```python
if (SENSORS is None):
        LOGGER.error(f"Failed to load sensors.yaml.")
```

drinfernoo, 2021-09-27:
Adding those lines, and then publishing to `wyzesense2mqtt/reload`... allowed the reload to work. Maybe the issue I was experiencing was a fluke, but it seemed like it would happen every time 🤔

Those lines shouldn't have affected the behavior of the script, from what I can tell... though there are a few lines lower that might fail if `SENSORS` *is* `None`:
https://github.com/raetha/wyzesense2mqtt/blob/944f9165a5346352ddd8300ca87a8ba68be42f4d/wyzesense2mqtt/wyzesense2mqtt.py#L181
https://github.com/raetha/wyzesense2mqtt/blob/944f9165a5346352ddd8300ca87a8ba68be42f4d/wyzesense2mqtt/wyzesense2mqtt.py#L206
But as you mentioned, I would think these would give an error.

My `sensors.yaml` is definitely formatted correctly... or none of my sensors would work at all 😉

drinfernoo, 2021-11-17:
I still have this odd behavior... wondering what we can do to fix this 🤔

raetha, 2025-03-03:
Tested this against the current build with all changes from @nabahr and reload seems to be working as expected. This area was changed with the addition of state persistence. Hopefully now fully resolved.

closed 2025-03-03 by raetha