raetha/wyzesense2mqtt #64: No communication since installing PCIE USB card

SkarnFlaxEnder, 2021-11-23:
Using a desktop and had wyzesense2mqtt fully functioning with several sensors connected. I shut it down to install a USB 3.0 card and on powering back up, auto detect no longer works for finding the bridge. It's been unable to detect any WyzeSense devices previously paired for a couple days now and have only just started troubleshooting.

Having the sensors integrated with HA, they stopped communicating as soon as the new card was installed, meaning it was on reboot. As this is the first time I've rebooted the machine since setup, I'm uncertain if it's related to the reboot or the new USB card having been installed. I did follow the steps for automatic startup during setup, as well as tried manually starting the service several times.

Currently running Pop OS 21.04 and the most recent pull of wyzesense2mqtt, don't know the version off the cuff but can edit it in later if needed.

The last several log entries are all identical, will exclude time stamps and IPs:

YYYY-MM-DD HH:MM:SS INFO     wyzesense2mqtt  Connecting to MQTT host ###.###.###.###
YYYY-MM-DD HH:MM:SS INFO     wyzesense2mqtt  Connecting to dongle auto
YYYY-MM-DD HH:MM:SS WARNING  wyzesense2mqtt  No device found on path auto: [Errno 2] No such file or directory: 'auto'

So far I've tried everything short of a complete uninstall-reinstall or directly hard coding the path into .yaml and rebooting to see if that fixes the issue. I'm about to try the hard coding the .yaml, but at a surface level it seems that adding new USB headers possibly breaks the auto?

SkarnFlaxEnder, 2021-11-23:
After hard coding the file path in config.yaml, it still didn't find the device. I then set it back to "auto" just to get back to baseline and it resolved itself after another reboot. Now I'm facing an issue where my wyzesense2mqtt is running on the same system as my HA VM instance, and it errors out as it can't open communications to my MQTT gateway which is integrated into HA on load... but I feel like that's not a bug, that's me setting it up poorly.

raetha, 2022-01-28:
@SkarnFlaxEnder, sorry for extreme delay, personal issues have kept me on the fringes of this project for most of the last year.

I'm not sure on what might have happened with the auto detection, it is based on how linux detects USB connected devices. It's possible with USB3 that something changed there, and maybe the extra reboot resolved it. I haven't explicitly done any testing with USB3 ports though, so this could happen again. You might check the linux system logs and device structure to make sure nothing odd is happening there, if the issue returns.

As for the communications side, I'm running all of my stuff extensively in docker, so can't fully help with VMs, but did have to resolve something similar. In docker I had to make sure that both my Mosquitto container and WS2MQTT were on the same virtual network in order to talk to each other. They could not route via the hosts network itself. So assuming you have both in VMs, try making sure they are on the same virtual switch, and possibly even jump into one of the VMs and make sure you can ping the other from it. You may have to access each by IP rather than hostname, if your VM stack doesn't handle automatic host resolution.

closed 2025-03-03 by raetha