raetha/wyzesense2mqtt #65: Disconnected from MQTT: out of memory

zee-shany, 2021-12-28:
Hello,

I've suddenly started to receive below errors, any one has any idea?

root@wyzesense2mqtt2:/etc/docker/wyzesense2mqtt/logs# tail wyzesense2mqtt.log
2021-12-28 09:54:14 INFO     wyzesense2mqtt  Disconnected from MQTT: Out of memory.
2021-12-28 09:54:14 INFO     wyzesense2mqtt  Disconnected from MQTT: The client is not currently connected.
2021-12-28 09:54:15 INFO     wyzesense2mqtt  Connected to MQTT: No error.
2021-12-28 09:54:15 ERROR    wyzesense2mqtt  failed to receive on socket: [Errno 104] Connection reset by peer
2021-12-28 09:54:15 INFO     wyzesense2mqtt  Disconnected from MQTT: Out of memory.
2021-12-28 09:54:15 INFO     wyzesense2mqtt  Disconnected from MQTT: The client is not currently connected.
2021-12-28 09:54:16 INFO     wyzesense2mqtt  Connected to MQTT: No error.
2021-12-28 09:54:16 ERROR    wyzesense2mqtt  failed to receive on socket: [Errno 104] Connection reset by peer
2021-12-28 09:54:16 INFO     wyzesense2mqtt  Disconnected from MQTT: Out of memory.
2021-12-28 09:54:16 INFO     wyzesense2mqtt  Disconnected from MQTT: The client is not currently connected.

I've tried restarting the VM where docker is running, Home assistant as well, unplug/plug the wyze USB dongle.

Thanks.

zee-shany, 2022-01-03:
hi,

Any help I can get here please? still receiving same error, i've even tried using another spare wyze USB dongle, updated the vm (ubnt 20) with latest updates. 

root@wyzesense2mqtt2:/etc/docker/wyzesense2mqtt/logs# tail wyzesense2mqtt.log
2022-01-03 10:47:07 INFO     wyzesense2mqtt  Disconnected from MQTT: Out of memory.
2022-01-03 10:47:07 INFO     wyzesense2mqtt  Disconnected from MQTT: The client is not currently connected.
2022-01-03 10:47:08 INFO     wyzesense2mqtt  Connected to MQTT: No error.
2022-01-03 10:47:08 ERROR    wyzesense2mqtt  failed to receive on socket: [Errno 104] Connection reset by peer
2022-01-03 10:47:08 INFO     wyzesense2mqtt  Disconnected from MQTT: Out of memory.
2022-01-03 10:47:08 INFO     wyzesense2mqtt  Disconnected from MQTT: The client is not currently connected.
2022-01-03 10:47:09 INFO     wyzesense2mqtt  Connected to MQTT: No error.
2022-01-03 10:47:09 ERROR    wyzesense2mqtt  failed to receive on socket: [Errno 104] Connection reset by peer
2022-01-03 10:47:09 INFO     wyzesense2mqtt  Disconnected from MQTT: Out of memory.
2022-01-03 10:47:09 INFO     wyzesense2mqtt  Disconnected from MQTT: The client is not currently connected.

appreciate your assistance.

raetha, 2022-01-28:
Hi @zee-shany. Sorry for delay, my personal life has drawn me away from smart home stuff for most of the last year unfortunately. Others have been still working on this project, but since I don't see anyone having jumped on I figured I'd take a quick look.

From the logging messages, it appears that your MQTT server is reporting that it is out of memory. From a quick search, that appears to be a message that shows up when it is being told to store messages with QOS 1 or 2. So my guess is that the disk space/memory allocated to your MQTT server isn't enough to store the number of persistent messages it is being asked to store. You could look at the resources allocated there and try to determine if that might be true. Alternately, WS2MQTT can't connect to MQTT, and it's docker container is running out of disk space/memory.

Those would be where I would start, unless you don't really need persistence, and then you could just change the WS2MQTT config file and set QOS to 0, which would eliminate any need to store the messages anywhere.

closed 2025-03-03 by raetha