raetha/wyzesense2mqtt #76: Version 3.0.1 not working in Unraid Docker

dchodoffHA, 2025-03-06:
## Describe the Bug

I upgraded to version and after the upgrade the program won't load. It just shuts off right away. 

In the logs get an error message saying - "cannot access local variable 'user_config' where it is not associated with a value"

I am using Unraid version 7.0.0 

When I downgrade back to v2.0 everything is working again. 

### Steps To Reproduce

Steps to reproduce the behavior:
Simply upgrading the version to 3.0.1 breaks the program and makes it unusable in Unraid. 

### Desktop

Please complete the following information:

- OS: Unraid 7.0.0
- Browser: Chrome
- Version: [e.g. 22]

## Additional Context

Add any other context about the problem here.

raetha, 2025-03-06:
Hi @dchodoffHA. I believe @nabahr's PR will improve the error handling here, but likely not solve your actual issue.

My best guess is that you missed the breaking changes section of the release notes for the 2.5+ versions I just built. I did some notable docker reworking and changed the volume mounts to use /app instead of /wyzesense2mqtt. This means in your docker command or compose file you need to update the mount points internal to the container to swap those values.

I'm going to accept the PR shortly, so please try again, just note that you won't get expected behavior without updating the volume as well. It just won't fail to start by using the sample config file instead.

raetha, 2025-03-06:
Ok, everything updated in devel branch. I'd appreciate if you could try the devel build/container before I publish the PR to the main branch.

dchodoffHA, 2025-03-06:
I just upgraded to devel branch and here is what I am getting (not working for me):

`docker run
  -d
  --name='wyzesense2mqtt'
  --net='bridge'
  --pids-limit 2048
  --privileged=true
  -e TZ="America/New_York"
  -e HOST_OS="Unraid"
  -e HOST_HOSTNAME="Tower"
  -e HOST_CONTAINERNAME="wyzesense2mqtt"
  -l net.unraid.docker.managed=dockerman
  -l net.unraid.docker.icon='https://github.com/digiblur/unraid-docker-templates/raw/master/images/wyzesense2mqtt.png'
  -v '/mnt/user/appdata/wyzesense2mqtt/config/':'/wyzesense2mqtt/config':'rw'
  -v '/mnt/user/appdata/wyzesense2mqtt/logs/':'/wyzesense2mqtt/logs/':'rw'
  --device='/dev/hidraw0' 'raetha/wyzesense2mqtt:devel'
3f525d6b1399cb9b9dd80877f7388ceaee419a34b3a3347a897a9a3a3398bc76`

Here is what the log is showing now: 
`
Logging initialized...
Initializing configuration...
Writing updated config file
/app/wyzesense2mqtt.py:143: DeprecationWarning: Callback API version 1 is deprecated, update to latest version
  MQTT_CLIENT = mqtt.Client(client_id=CONFIG['mqtt_client_id'], clean_session=CONFIG['mqtt_clean_session'])
Connecting to MQTT host <host>`

The sensors are not responding in the logs or in HA (via MQTT)

dchodoffHA, 2025-03-06:
@raetha I was going through the code changes and just noticed my mistake. 

Just updated the paths and now it's working! 

Thanks so much for keeping this project alive!

raetha, 2025-03-06:
Awesome! Glad it really was that simple. I'm still going to see if a symlink will work for backwards compatibility, as I'm sure others who realize this is alive again might miss the same thing. Generally I wouldn't have made major changes, but I'm trying new standards with a few of my docker projects for consistency. :)

raetha, 2025-03-06:
Ok, I can't figure out how to allow bindmounts via a symlink folder. So I think there's no easy way to make it backwards compatible. I'll leave this open for a bit in case anyone who sees it knows a way to possibly do this.

closed 2025-03-08 by raetha