raetha/wyzesense2mqtt #85: Temp Sensor report Fahrenheit not Celsius

jfarmer08, 2025-07-27:
After I manually change this to Fahrenheit its now reporting correctly in HA.

**After Change**
`{"name": "Temperature", "device_class": "temperature", "state_class": "measurement", "unit_of_measurement": "\u00b0C", "json_attributes_topic": "wyzesense2mqtt/77D53275", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "manufacturer": "Wyze", "model": "WyzeSense Climate Sensor", "name": "WyzeSense 77D53275", "sw_version": "21", "via_device": "wyzesense2mqtt"}, "value_template": "{{ value_json.temperature }}", "unique_id": "wyzesense_77D53275_temperature", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
  homeassistant/sensor/wyzesense_77D53275/humidity/config
  {"name": "Humidity", "device_class": "humidity", "state_class": "measurement", "unit_of_measurement": "%", "json_attributes_topic": "wyzesense2mqtt/77D53275", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "name": "WyzeSense 77D53275"}, "value_template": "{{ value_json.humidity }}", "unique_id": "wyzesense_77D53275_humidity", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
  homeassistant/sensor/wyzesense_77D53275/signal_strength/config
  {"device_class": "signal_strength", "state_class": "measurement", "unit_of_measurement": "dBm", "entity_category": "diagnostic", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "name": "WyzeSense 77D53275"}, "value_template": "{{ value_json.signal_strength }}", "unique_id": "wyzesense_77D53275_signal_strength", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
  homeassistant/sensor/wyzesense_77D53275/battery/config
  {"device_class": "battery", "state_class": "measurement", "unit_of_measurement": "%", "entity_category": "diagnostic", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "name": "WyzeSense 77D53275"}, "value_template": "{{ value_json.battery }}", "unique_id": "wyzesense_77D53275_battery", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
State event data: sensor_type=climate,battery=100,signal_strength=-54,temperature=85.45,humidity=45,event=status,mac=77D53275,timestamp=1753636479.982
wyzesense2mqtt/77D53275
{'class': 'temperature', 'name': 'WyzeSense 77D53275', 'sensor_type': 'climate', 'sw_version': '21', 'unit_of_measurement': '°F', 'invert_state': False, 'battery': 100, 'signal_strength': -54, 'temperature': '85.45', 'humidity': 45, 'event': 'status', 'mac': '77D53275', 'timestamp': 1753636479.982}
State event data: sensor_type=climate,battery=100,signal_strength=-39,temperature=86.36,humidity=42,event=status,mac=77D53275,timestamp=1753636510.473
wyzesense2mqtt/77D53275
{'class': 'temperature', 'name': 'WyzeSense 77D53275', 'sensor_type': 'climate', 'sw_version': '21', 'unit_of_measurement': '°F', 'invert_state': False, 'battery': 100, 'signal_strength': -39, 'temperature': '86.36', 'humidity': 42, 'event': 'status', 'mac': '77D53275', 'timestamp': 1753636510.473}
Logging initialized...`

https://github.com/raetha/wyzesense2mqtt/blob/6e922545ea6c07a9b374421a95c51c4f1f3c7181/wyzesense2mqtt/wyzesense2mqtt.py#L516C38-L516C39

**Before Change **
Logs from Sensor `{"name": "Temperature", "device_class": "temperature", "state_class": "measurement", "unit_of_measurement": "\u00b0C", "json_attributes_topic": "wyzesense2mqtt/77D53275", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "manufacturer": "Wyze", "model": "WyzeSense Climate Sensor", "name": "WyzeSense 77D53275", "sw_version": "21", "via_device": "wyzesense2mqtt"}, "value_template": "{{ value_json.temperature }}", "unique_id": "wyzesense_77D53275_temperature", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
  homeassistant/sensor/wyzesense_77D53275/humidity/config
  {"name": "Humidity", "device_class": "humidity", "state_class": "measurement", "unit_of_measurement": "%", "json_attributes_topic": "wyzesense2mqtt/77D53275", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "name": "WyzeSense 77D53275"}, "value_template": "{{ value_json.humidity }}", "unique_id": "wyzesense_77D53275_humidity", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
  homeassistant/sensor/wyzesense_77D53275/signal_strength/config
  {"device_class": "signal_strength", "state_class": "measurement", "unit_of_measurement": "dBm", "entity_category": "diagnostic", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "name": "WyzeSense 77D53275"}, "value_template": "{{ value_json.signal_strength }}", "unique_id": "wyzesense_77D53275_signal_strength", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
  homeassistant/sensor/wyzesense_77D53275/battery/config
  {"device_class": "battery", "state_class": "measurement", "unit_of_measurement": "%", "entity_category": "diagnostic", "device": {"identifiers": ["wyzesense_77D53275", "77D53275"], "name": "WyzeSense 77D53275"}, "value_template": "{{ value_json.battery }}", "unique_id": "wyzesense_77D53275_battery", "state_topic": "wyzesense2mqtt/77D53275", "availability": [{"topic": "wyzesense2mqtt/77D53275/status"}, {"topic": "wyzesense2mqtt/status"}], "availability_mode": "all", "platform": "mqtt"}
State event data: sensor_type=climate,battery=100,signal_strength=-54,temperature=85.45,humidity=45,event=status,mac=77D53275,timestamp=1753636479.982
wyzesense2mqtt/77D53275
{'class': 'temperature', 'name': 'WyzeSense 77D53275', 'sensor_type': 'climate', 'sw_version': '21', 'unit_of_measurement': '°F', 'invert_state': False, 'battery': 100, 'signal_strength': -54, 'temperature': '85.45', 'humidity': 45, 'event': 'status', 'mac': '77D53275', 'timestamp': 1753636479.982}
State event data: sensor_type=climate,battery=100,signal_strength=-39,temperature=86.36,humidity=42,event=status,mac=77D53275,timestamp=1753636510.473
wyzesense2mqtt/77D53275
{'class': 'temperature', 'name': 'WyzeSense 77D53275', 'sensor_type': 'climate', 'sw_version': '21', 'unit_of_measurement': '°F', 'invert_state': False, 'battery': 100, 'signal_strength': -39, 'temperature': '86.36', 'humidity': 42, 'event': 'status', 'mac': '77D53275', 'timestamp': 1753636510.473}
Logging initialized...`

raetha, 2025-07-28:
Hi @jfarmer08, just pushed an update to the devel branch that switches to fahrenheit for the unit of measurement. Can you pull the updated version and confirm that it is working for you as expected? Thanks for reporting this, I don't have a climate sensor to test with so that code was entirely from HclX and may result in that needing to be a configuration option to fully support all users.

jfarmer08, 2025-08-03:
Works fine with that.