raetha/wyzesense2mqtt #87: Home assistant addon.

jfarmer08, 2025-08-03:
I am in the process of creating a HA addon using this repo. Are you willing to making changes as needed to support it 

https://github.com/jfarmer08/home-assistant-addons

jfarmer08, 2025-08-04:
What I’ve done to use your Docker image in Home Assistant
Created a Home Assistant addon using your raetha/wyzesense2mqtt:devel Docker image.
Automated config file generation (config.yaml and sensors.yaml) from addon options, including support for inline YAML sensor lists in the UI.
Ensured the container exposes /dev/hidraw0 for USB dongle access.
Added schema validation and grouped config options for a clean Home Assistant UI.
Provided full documentation and troubleshooting steps in the addon info tab, including clickable links and images.
Made the addon installable via Supervisor Add-on Store with a custom repository.
Suggestions for improvements to the Docker image
Config File Flexibility:
Allow config and sensor files to be provided via environment variables or mounted paths, so users don’t need to manually edit files inside the container.

Sensor List Format:
Support both inline YAML lists and standard YAML dictionaries for sensors, making it easier to generate configs from UI or scripts.

Logging and Error Output:
Improve error messages for config issues (e.g., invalid YAML, missing fields) and log them clearly to stdout/stderr for easier debugging in Home Assistant.

Graceful Reload:
Add a command or API endpoint to reload config/sensors without restarting the container, for smoother Home Assistant integration.

Documentation:
Include a section in your README about Home Assistant addon compatibility and best practices for config file mounting and sensor setup.

Image Size:
Consider using a minimal base image and multi-stage builds to reduce image size and speed up downloads.

Additional Recommendations
Environment Variable Support:
Allow all major configuration options (MQTT host, port, credentials, topic roots, etc.) to be set via environment variables, not just config files. This is the standard for most containerized apps and makes integration with Home Assistant and other platforms much easier.

Config File Override Logic:
If both environment variables and config files are present, document the precedence (e.g., env vars override config file values).

Automatic Sensor Config Discovery:
If a sensor config file is not provided, auto-generate one from environment variables or a default template, reducing manual steps.

Healthcheck Endpoint:
Add a simple HTTP or MQTT healthcheck endpoint so Home Assistant can monitor the container’s status.

Graceful Shutdown and Reload:
Ensure the service handles SIGTERM/SIGHUP for clean shutdowns and config reloads, which is important for container orchestration.

Config Validation on Startup:
Validate config and environment variables at startup, and log clear errors if anything is missing or invalid.

Documentation Update:
Add a section to your README about best practices for running in containers, including environment variable usage and Home Assistant integration tips.

raetha, 2025-11-09:
Hi @jfarmer08! I'm very happy to support this, and it was on my roadmap, I just wasn't sure how many people were still using this. Not sure if the source repo can also be the HA AddOn repo with added files, but if so feel free to do a PR adding those files here. If not, let me know what you need, or submit PR's with changes. Just please submit against the devel branch as I start everything there for some quick testing before promoting.